intel/media-driver #4: Compile error c++ (GCC) 7.2.1 20170915 (not a constant expression)

uartie, 2017-12-01:
With `c++ (GCC) 7.2.1 20170915` (i.e. Fedora 27 default GCC version), the following compiler error is generated:
```
In file included from <snip>/media_driver/agnostic/common/os/mos_context.cpp:28:0:
<snip>/media_driver/linux/common/os/mos_context_specific.h:63:67: error: value ‘18446744073709551615’ of type ‘void*’ is not a constant expression
     constexpr static void* MOS_LINUX_SHM_INVALID       = (void *)-1;
                                                                   ^
```

uartie, 2017-12-01:
This is also the default compiler version on Fedora 26.

On Fedora 25 (which is EOL mid-December 2017), the compiler version is 6.4.1 and this error does not occur.

mkslack, 2017-12-02:
Same issue with SlackwareLinux /current:

Reading specs from /usr/lib64/gcc/x86_64-slackware-linux/7.2.0/specs
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-slackware-linux/7.2.0/lto-wrapper
Target: x86_64-slackware-linux
Configured with: ../gcc-7.2.0/configure --prefix=/usr --libdir=/usr/lib64 --mandir=/usr/man --infodir=/usr/info --enable-shared --enable-bootstrap --enable-languages=ada,brig,c,c++,fortran,go,lto,objc --enable-threads=posix --enable-checking=release --enable-objc-gc --with-system-zlib --enable-libstdcxx-dual-abi --with-default-libstdcxx-abi=gcc4-compatible --disable-libunwind-exceptions --enable-__cxa_atexit --enable-libssp --enable-lto --disable-install-libiberty --with-gnu-ld --verbose --with-arch-directory=amd64 --disable-gtktest --disable-multilib --target=x86_64-slackware-linux --build=x86_64-slackware-linux --host=x86_64-slackware-linux
Thread model: posix
gcc version 7.2.0 (GCC)

ln421, 2017-12-04:
Thanks uartie for comments. Currently intel Linux driver is built against ubuntu 16.04 w/ gcc 5.4.  Gcc 7.2 definitely implement stricter check for constant declared as constexpr. Will fix the issue and set up Fedora 26 environment for cross check soon. Please stay tuned.

PR #20 (2017-12-26): Fix mos build issue
fix #4
closed 2017-12-26 by oliver-sang