intel/media-driver #7: VAEncMiscParameterBufferQualityLevel need to reset for AVC/HEVC when sequence parameters reset.

mypopydev, 2017-12-04:
User need to reset  VAEncMiscParameterBufferQualityLevel after reset sequence parameters , otherwise iHD driver won't use the expected quality level for encoding.

BTW:  iHD driver uses a different quality level for AVC/HEVC when reset, AVC use 1, HEVC use 4. 

I don't think this action follow the VA-API Spec, and iHD follow a  counterintuitive way for Quality Level setting.

xinfengz, 2017-12-04:
you mean, even there are a new sequence parameters, driver should always keep previous quality level setting until next quality level setting, right?   so , we should not set the default quality level in parsing sequence parameter function?

mypopydev, 2017-12-04:
Yes, I think quality level will keep  unless user explicit change it, if user does not debug the driver, do you think any user know need to re-setting quality level when re-set sequence parameter? 

BTW: AVC/HEVC use a different default quality level value (1 for AVC, 4 for HEVC), I don't think this is a good idea from API view.

xinfengz, 2017-12-04:
sure , will submit a PR to fix it . thanks

PR #56 (2018-01-31): keep quality level until new VAEncMiscParameterBufferQualityLevel
fix #7
commit ea892df3 (2018-01-31): keep quality level until new VAEncMiscParameterBufferQualityLevel
fix #7
commit 5fe0bde0 (2018-01-31): keep quality level until new VAEncMiscParameterBufferQualityLevel (CL#734426)

fix #7, PR #56

Change-Id: I635d509c2b7066ffd71018b2ec3faf9908bff6a1
closed 2018-01-31 by oliver-sang