intel/media-driver #9: vaQuerySurfaceAttributes doesn't support ENC

mypopydev, 2017-12-04:
vaQuerySurfaceAttributes support DEC and VPP, but doesn't support ENC

xinfengz, 2017-12-06:
yes, not implemented for encoder, I wonder whether it is  "MUST SUPPORT"  , thanks

mypopydev, 2017-12-07:
If user want to query ENC some information (e,g,  input pixel format, max size ... ) ,  vaQuerySurfaceAttributes is the only interface for this type query, and if ENC change some thing (e,g, maybe support YUV422 in feature， change the max size from 4K to 16K...), no other way to query the ENC, so I think it is "MUST SUPPORT".

xinfengz, 2017-12-07:
input pixel format can be covered by VAConfigAttribRTFormat, max encode resolution can be supported by VAConfigAttribMaxPictureWidth VAConfigAttribMaxPictureHeight. vaQuerySurfaceAttributes can be used for surface attribute. for example: surface maxium width can be larger than encode ability resolution. they are different things.

mypopydev, 2017-12-08:
Yes, when encoding, we need to use vaQuerySurfaceAttributes to get the surface constraints, and I don't think VAConfigAttribRTFormat can cover precise pixel format, e,g VA_RT_FORMAT_YUV420 means YUV420P or NV12?

XinfengZhang, 2017-12-16:
please help to review  #29 , thanks

closed 2018-01-11 by XinfengZhang