intel/media-driver #16: ~2000 test failures in test_va_api

uartie, 2017-12-07:
`test_va_api` is part of https://github.com/01org/libva-utils and is used for simple VA API conformance validation.  Currently, when executed with `LIBVA_DRIVER_NAME=iHD` there are ~2000 test failures.  Some test failures are inevitable (since they may be specific to i965 driver), but many others are real failures.

FYI... i965 driver has 0 failures in `test_va_api` as of writing.

uartie, 2017-12-07:
#12 and #15 are related to some of these failures

xinfengz, 2017-12-07:
some test in libva-utils is not correct. such as test_va_api_fixture.cpp:117, why the value must is 32, it should be 38 with current libva master

uartie, 2017-12-07:
> some test in libva-utils is not correct. such as test_va_api_fixture.cpp:117, why the value must is 32, it should be 38 with current libva master

Please test with latest libva-utils master... this was fixed for about a week now (https://github.com/01org/libva-utils/commit/597541534c36526f9674b032b21e1a84e43e9d8d).  There are also various other fixes to libva-utils to make it more driver agnostic.

If you discover other issues with the tests themselves... can you kindly file a new issue in the libva-utils project?  I am also attempting to identify/fix some of the failures as time permits.

XinfengZhang, 2017-12-26:
now, still 1689 failed cases. will continue, it should be a long term task.

XinfengZhang, 2017-12-26:
after merging PR 56 , still 1450 failure exist.

XinfengZhang, 2017-12-28:
when call vaCreateConfig for encode. if VAConfigAttribRateControl is not set, is "not supported" expected.

XinfengZhang, 2017-12-29:
after #66 & #61 , still 1072 failed tests

XinfengZhang, 2017-12-29:
https://github.com/01org/libva-utils/issues/83
this libva-utils issue also caused some failed

XinfengZhang, 2018-03-20:
close this issue and trace with new issue #120

closed 2018-03-20 by XinfengZhang