intel/media-driver #17: vaGet/SetDisplayAttributes returns VA_STATUS_ERROR_UNIMPLEMENTED

uartie, 2017-12-07:
Will this get implemented?

Also relates to #16

xinfengz, 2017-12-07:
are there are real usage case? who are using these interface?

xinfengz, 2017-12-07:
this is a pure video codec/vp driver, I wonder whether it is MUST HAVE feature for a codec driver.

uartie, 2017-12-07:
That's a good question.  IDK.  Perhaps @xhaihao or @chivakker know of specific usage cases.

Nonetheless, if it's not going to be implemented then `ctx->max_display_attributes` (i.e. vaMaxNumDisplayAttributes) should be 0?  Currently it returns 4.

uartie, 2017-12-07:
From a quick `grep` on a few source trees, I see that gstreamer-vaapi, libyami, and mesa use these API's.

chivakker, 2017-12-07:
it is a supported API and also needed.  You'll be breaking all mddleware basic checks which will result in failure.  All opensource middleware uses it.

xinfengz, 2017-12-08:
if I implement the interface,but not real support. for example: still not support Brightness,contrast ect tuning. is it ok?

fhvwy, 2017-12-08:
From my point of view, the PutSurface and related interfaces only exist for testing - they aren't sufficiently adaptable to be used for proper playback, and the few display attributes being considered here add little to that.  VAAPI is meant for codecs, and the display path should be left to APIs actually intended for it - nothing in ffmpeg/libav uses these interfaces, nor do the players based on it (vlc, mpv).

So, certainly it shouldn't return that display attributes are present when they aren't, but I don't think anything beyond that is needed for a codec driver.  Compare the Mesa implementation, which is essentially the same as what you already have: <https://cgit.freedesktop.org/mesa/mesa/tree/src/gallium/state_trackers/va/display.c>.

(Do implement it of course if Intel has specific requirements, but the comment "All opensource middleware uses it." is not true.)

ceyusa, 2017-12-08:
As far as I understand, depends on the driver to decide what to implement, and the users should be able to query the driver to know what can be done.

Said that, I was thinking that if the driver doesn't implement anything related with displaying surfaces it perhaps the function vaMaxNumDisplayAttributes() should return -1

But that also means that gstreamer-vaapi should be changed to bail out early vaapisink or perhaps even not register it at all.

XinfengZhang, 2017-12-13:
I suppose middle ware can handle the VA_STATUS_ERROR_UNIMPLEMENTED , if return unimplemented , it should be same as vaMaxNumDisplayAttributes == -1, right?

uartie, 2017-12-14:
I'm not sure how various middleware handles the vaMaxNumDisplayAttributes result.  However, I suspect a situation where a poorly written middleware might pass-through the result of vaMaxNumDisplayAttributes to malloc unconditionally to create a list of VADisplayAttribute.  The malloc takes a "size_t" (i.e. unsigned) and if middleware passes -1 to malloc, then it could be quite fatal.  Seems 0 return value for vaMaxNumDisplayAttributes might be safer.

uartie, 2017-12-14:
... that said, seems LibVA's vaMaxNum/Get/SetDisplayAttributes API's declare the wrong type (int) for num_attributest to begin with.

XinfengZhang, 2017-12-14:
yes, -1 is not safe, set it to zero is suitable , but I wonder why middle ware does not handle VA_STATUS_ERROR_UNIMPLEMENTED ?

XinfengZhang, 2017-12-14:
is #28 is ok ?

uartie, 2017-12-14:
yes, LGTM.

uartie, 2017-12-14:
> ...but I wonder why middle ware does not handle VA_STATUS_ERROR_UNIMPLEMENTED ?

I don't think that's the real issue.  If middleware is not handling that, then it is a bug in middleware.  But, as @fhvwy pointed out:

> So, certainly it shouldn't return that display attributes are present when they aren't...

Basically, the driver is telling vaapi apps/middleware that there are 4 supported display attributes but then does not implement them... that could confuse/break apps/middleware.  Therefore, #28 should be sufficient to address this.

XinfengZhang, 2017-12-17:
there are related code in libva, CHECK_MAXIMUM(vaStatus, ctx, display_attributes);, if max_display_attributes ==0, it will return error.

uartie, 2017-12-18:
@xhaihao, what's your recommendations?

xhaihao, 2017-12-20:
Please submit a PR to fix libva @uartie @XinfengZhang

closed 2018-02-28 by XinfengZhang