intel/media-driver #21: VC1DEC:  b_picture_fraction not follow VA-API 2.0 define

mypopydev, 2017-12-11:
in va.h the b_picture_fraction in va.h means Index for PICTURE_LAYER::BFRACTION value in Table 40 (7.1.1.14), but media-driver not follow the define

thomasli21801, 2017-12-26:
The parse function is following VC1 bspec. In order to follow definition in va.h, app must make sure the calculation to map code word to index of FractionToScaleFactor table. Could you please help to double check with app behavior?

mypopydev, 2017-12-28:
@thomasli21801 in va.h, b_picture_fraction means index for PICTURE_LAYER::BFRACTION value in Table 40 (7.1.1.14), please double check the define. Tks

thomasli21801, 2017-12-28:
We may need to modify internal test app for VC1 decode to change from code word to index.

closed 2018-02-28 by XinfengZhang