intel/media-driver #25: libva-utils ENC test samples avcenc can't work.

mypopydev, 2017-12-13:
https://github.com/01org/libva-utils/blob/master/encode/avcenc.c can't work in media-driver, the root case is setting 0/0 in    /* Create a context for this decode pipe */
    va_status = vaCreateContext(va_dpy, avcenc_context.config_id,
                                picture_width, picture_height,
                                VA_PROGRESSIVE, 
                                0, 0,
                                &avcenc_context.context_id);

XinfengZhang, 2017-12-13:
driver store the surfaceID array to be reference or reconstruct. all recon/reference should be in the stored surfaceID array. can change it to dynamic register the surface, but is it very important?  I can evaluate the priority

mypopydev, 2017-12-14:
@XinfengZhang I observe FEI didn't follow this restrict, any  specialized reason?

XinfengZhang, 2017-12-14:
I implemented it by requirement from real usagecases,only decode and FEI.  we will add the support for legacy encode.

XinfengZhang, 2018-01-16:
fixed by  the commit 039e892

closed 2018-01-16 by XinfengZhang