intel/media-driver #32: Provide support for cmake 2.8 and align cmake requirements across sources

dvrogozh, 2017-12-18:
Some Linux distributions, especially server distributions, still has older cmake versions than 3.5 by default. Cent OS 7.4 is an example:
```
# cmake --version
cmake version 2.8.12.2
```
Yocto is another example:
```
# cmake --version
cmake version 3.3.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
Please, provide compatibility with older versions of cmake starting from 2.8 to extend Linux distributions coverage.

Also, please, align cmake version requirements across sources. As it stands now:
```
# fgrep -rsn cmake_minimum_required .
./Tools/MediaDriverTools/GenKrnBin/CMakeLists.txt:21:cmake_minimum_required(VERSION 2.6)
./Tools/MediaDriverTools/KernelBinToSource/CMakeLists.txt:21:cmake_minimum_required (VERSION 2.8)
./Tools/MediaDriverTools/KrnToHex_IGA/CMakeLists.txt:21:cmake_minimum_required(VERSION 2.6)
./cmrtlib/CMakeLists.txt:22:cmake_minimum_required(VERSION 2.8)
./cmrtlib/linux/CMakeLists.txt:21:cmake_minimum_required(VERSION 2.8)
./media_driver/CMakeLists.txt:22:cmake_minimum_required(VERSION 3.5)
./CMakeLists.txt:21:cmake_minimum_required(VERSION 3.5)
```

Mind that I have submitted similar request against gmmlib: https://github.com/intel/gmmlib/issues/1

oliver-sang, 2018-01-29:
though [intel/gmmlib#1](https://github.com/intel/gmmlib/issues/1) has been rejected, media will keep this issue open.
we have plan to decouple the media/gmmlib build in the future. At that time we will re-evaluate this cmake version request. Thanks

XinfengZhang, 2020-06-23:
#939

XinfengZhang, 2023-03-23:
the issue still there, previously, @dvrogozh tried 2.8.12 and it works. 
now, there are more different requirements, 
from 2.6 to 3.5, need to unify the requirement.  @MicroYY , could you help to check media-driver minimum requirement to keep maximum convenience to user?

MicroYY, 2023-03-24:
The minimum required cmake version is 3.4, as cmake function `pkg_get_variable` is introduced in cmake 3.4.
https://github.com/intel/media-driver/blob/1390e5e905c85fb40bcd118550d848d2042335db/media_driver/cmake/linux/media_include_paths_linux.cmake#L33