intel/media-driver #34: MJPEG ENC return a lot of attributes, but the iHD didn't support them actually

mypopydev, 2017-12-19:
For MJPEG ENC, iHD return a lot attributes when call vaGetConfigAttributes(), but the driver can't support them, I think if the driver can't support the attributes, need to return value  VA_ATTRIB_NOT_SUPPORTED, now iHD didn't follow this action.

(gdb) p *attributeList 
$14 = std::map with 21 elements = {[VAConfigAttribRTFormat] = 196639, [VAConfigAttribEncryption] = 2147483648, [VAConfigAttribRateControl] = 150, [VAConfigAttribEncPackedHeaders] = 2147483648, 
  [VAConfigAttribEncInterlaced] = 0, [VAConfigAttribEncMaxRefFrames] = 65537, [VAConfigAttribEncMaxSlices] = 0, [VAConfigAttribEncSliceStructure] = 2, [VAConfigAttribMaxPictureWidth] = 16384, 
  [VAConfigAttribMaxPictureHeight] = 16384, [VAConfigAttribEncJPEG] = 53440, [VAConfigAttribEncQualityRange] = 1, [VAConfigAttribEncQuantization] = 0, [VAConfigAttribEncIntraRefresh] = 0, 
  [VAConfigAttribEncSkipFrame] = 1, [VAConfigAttribEncROI] = 0, [VAConfigAttribProcessingRate] = 1, [VAConfigAttribEncDirtyRect] = 4, [VAConfigAttribEncParallelRateControl] = 1, 
  [VAConfigAttribFEIMVPredictors] = 0, [VAConfigAttribCustomRoundingControl] = 0}

And I think **VAStatus MediaLibvaCaps::CreateEncAttributes()** need more strict check for other CODEC

XinfengZhang, 2017-12-20:
yes, we should remove the skip frame, processingrate etc.

XinfengZhang, 2018-01-11:
@mypopydev if there are such attributes,but the value is disable/invalid. is it acceptable? for example,roi value is 0, means non surport

uartie, 2018-01-11:
When an attribute is not supported by profile/entrypoint, the value must be set to VA_ATTRIB_NOT_SUPPORTED by driver in vaGetConfigAttributes.  See https://github.com/01org/libva/blob/22eecc54131e5c3a4d54013511d98d01ce53cef9/va/va.h#L1037

mypopydev, 2018-01-11:
I think VA_ATTRIB_NOT_SUPPORTED more better in this case,  or else the caller need to handle VA_ATTRIB_NOT_SUPPORTED and XXX value == 0 or the other value means not supportted, it's ugly.

uartie, 2018-01-11:
Also see documentation for vaGetConfigAttributes... https://github.com/01org/libva/blob/22eecc54131e5c3a4d54013511d98d01ce53cef9/va/va.h#L1083

XinfengZhang, 2018-01-11:
thanks, I agree, set it to VA_ATTRIB_NOT_SUPPORTED is more suitable