intel/media-driver #35: Consider git submodule for gmmlib dependency

uartie, 2017-12-19:
Since commit intel/gmmlib@aa11d6ada523 , the media-driver does not compile (see intel/gmmlib#3) on current media-driver master (4d890718d862eb2325ded6aba18ecfcc15ce2036).

If gmmlib was included as a `git submodule` in media-driver, then the specific commit can be controlled more precisely and prevent recent changes in gmmlib from breaking the media-driver when not warranted.  This would also allow media-driver to remove all the cmake scripts that it currently duplicates from gmmlib to simplify maintainability (e.g. keeping them in sync is currently a chore).

uartie, 2017-12-19:
Curious, is there a reason for separating gmmlib source tree from media-driver source tree (e.g. is it used by other libraries/projects)?  Also, since it's separated, why does it not provide standard installation files?  The current build setup rules are awkward and unconventional because of this.

oliver-sang, 2017-12-19:
the corresponding media part changes are under testing now. We'll push the changes when tests pass.

Yes, gmmlib is a common component which is now built as a static library then link into different gmmlib 'clients'. There are some tasks for gmmlib changes in the future which should be able to address your concern. Thanks for pointing out these for us now!

oliver-sang, 2017-12-19:
fixed at 4bff0eba01d78896dafa7682d60b55d3cd8f839b for build break

drprajap, 2017-12-19:
We have task under progress to transition from static lib to DLL. We should be able to control such issues once transition happens. 

Sorry for any inconvenience. 
Thanks.

chivakker, 2018-07-20:
gmmlib is ready for this to happen, will it be available soon?

oliver-sang, 2018-07-24:
team is working on this.

dvrogozh, 2018-07-24:
Folks, I very much don't like an idea to have gmmlib as a submodule assuming that it will be a dynamic library. In that way we will end up having few versions of dynamic gmmlib built within media-driver and compute-runtime procedures. Imho, if gmmlib will be a dynamic library it should be separately installed and packaged. If it will be a static library, then indeed we can have a submodule.

@oliver-sang : can you, please, clarify what team is actually heading for?

oliver-sang, 2018-07-25:
Team is working to convert to use dynamic gmmlib. git submodule is not under the consideration.

XinfengZhang, 2019-08-27:
close it , because gmmlib is a .so now.

closed 2019-08-27 by XinfengZhang