intel/media-driver #38: [gstreamer-vaapi-master][iHD_SKL] [AVC_decode] "h264parse0: Internal data stream error" error message occurs during run h264 decoding cases with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. take one case for example, run command
gst-launch-1.0 -q filesrc location=/media//h264/CAPCM1_Sand_E.264 ! h264parse ! vaapih264dec ! vaapipostproc format=i420 ! checksumsink2 frame-checksum=FALSE file-checksum=TRUE plane-checksum=FALSE
3. error info. as below
0:00:00.041966159  2891      0x1520e30 ERROR       vaapivideomemory gstvaapivideomemory.c:278:map_vaapi_memory: failed to make image current
0:00:00.041992987  2891      0x1520e30 ERROR                default video-frame.c:161:gst_video_frame_map_id: failed to map video frame plane 0
0:00:00.042020944  2891      0x1520e30 ERROR          checksumsink2 gstchecksumsink.c:381:gst_cksum_image_sink_render:<cksumimagesink0> failed to map frame
ERROR: from element /GstPipeline:pipeline0/GstH264Parse:h264parse0: Internal data stream error.
Additional debug info:
gstbaseparse.c(3597): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH264Parse:h264parse0:
streaming stopped, reason error (-5)
d41d8cd98f00b204e9800998ecf8427e

XinfengZhang, 2017-12-20:
are there vaapi level log? which vaapi failed? it can help us to quick check.

Tianhaol, 2018-01-11:
An I420 surface is created by calling gst_vaapi_surface_create_full with attribute list in which I420 format is assigned, the corresponding VAImage is created by gst_vaapi_image_create, unfortunately VAImage with I420 format is not supported,  I420 is convert to YV12 in this function,  so the YV12 image is created. When call ensure_image_is_current in gst_vaapi,  then call DdiMedia_GetImage of driver, because  surface format is not compatible with the image,  error is returned from driver.

XinfengZhang, 2018-01-19:
@wangzj0601 please check why you create I420 surface, suppose you can get the surface format by vaQueryConfigAttributes.

closed 2018-04-03 by XinfengZhang