intel/media-driver #40: [gstreamer-vaapi-master][iHD_SKL] [VC1_decode] "filesrc0: Internal data stream error" error message occurs during run VC1 decode cases with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. take one case for example, run command
gst-launch-1.0 -q filesrc location=/media//yami_vc1/SA10086.vc1 '!' 'video/x-wmv,profile=(string)advanced,' width=720, height=576, framerate=14/1 '!' vaapivc1dec '!' vaapipostproc format=i420 '!' checksumsink2 frame-checksum=FALSE file-checksum=TRUE plane-checksum=FALSE
3. error info. as below
0:00:00.045216413  4074      0x189acf0 ERROR       vaapivideomemory gstvaapivideomemory.c:278:map_vaapi_memory: failed to make image current
0:00:00.045252735  4074      0x189acf0 ERROR                default video-frame.c:161:gst_video_frame_map_id: failed to map video frame plane 0
0:00:00.045320055  4074      0x189acf0 ERROR          checksumsink2 gstchecksumsink.c:381:gst_cksum_image_sink_render:<cksumimagesink0> failed to map frame
ERROR: from element /GstPipeline:pipeline0/GstFileSrc:filesrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:pipeline0/GstFileSrc:filesrc0:
streaming stopped, reason error (-5)
d41d8cd98f00b204e9800998ecf8427e

Tianhaol, 2018-01-12:
Root Cause: Same as https://github.com/intel/media-driver/issues/38

XinfengZhang, 2018-04-03:
fixed by #113

closed 2018-04-03 by XinfengZhang