intel/media-driver #43: [gstreamer-vaapi-master][iHD_SKL] [JPEG_decode] 52 cases decoding fail during run jpeg decoding cases with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. take one case for example, run command
gst-launch-1.0 filesrc location=/media//JPEG_RefCompare/400_f_14312182_t8o4hv9r9eqs17mc1zzykfaxeufdh0p0.jpg '!' jpegparse '!' vaapijpegdec '!' videoconvert '!' video/x-raw, format=I420 '!' filesink location=400_f_14312182_t8o4hv9r9eqs17mc1zzykfaxeufdh0p0.jpg.yuv
3. error info. as below
0:00:00.034652253 17348       0xf33ed0 ERROR            vaapidecode gstvaapidecode.c:716:gst_vaapidecode_handle_frame: decode error -1
ERROR: from element /GstPipeline:pipeline0/GstVaapiDecode_jpeg:vaapidecode_jpeg0: No valid frames decoded before end of stream

Failed cases
decode_jpeg/126296226_bf7eba45c21.jpg yuvj444p 500 332    DecodeFAIL
decode_jpeg/127746-fb.jpg yuvj444p 338 450    DecodeFAIL
decode_jpeg/1-gbb-peace-389.jpg yuvj444p 600 464    DecodeFAIL
decode_jpeg/29-4673tourists-rice-boat-on-the-backwaters-near-kayamkulam-kerala-india-posters.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/29-725.jpg yuvj444p 400 296    DecodeFAIL
decode_jpeg/3245_by_chris_johansen-thumb.jpg yuvj444p 500 656    DecodeFAIL
decode_jpeg/34tew40.jpg yuvj444p 450 734    DecodeFAIL
decode_jpeg/390-2377.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/400_f_11014777_vbypseclj6jfkjhgn5wsiidtrm5j0ukm.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/400_f_11333253_dkuzd0h1xj8oqtwsxz9ea9nujxj5mjrq.jpg yuvj444p 400 320    DecodeFAIL
decode_jpeg/400_f_13944420_gihrb2varxosawnjrvflrggxvnpmkeyg.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_14004535_9qiymls1kpn2mq8kgdfysndvw4qimbtl.jpg yuvj444p 400 296    DecodeFAIL
decode_jpeg/400_f_14150758_epe7fh0blmqvoxqstyzff9kz0xxjyt9t.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/400_f_14201614_jqrgiltby92rwjadydtqlagod5tc6njg.jpg yuvj444p 400 314    DecodeFAIL
decode_jpeg/400_f_14310896_gaep1uru4nutibf98qt4iicukm0vayeg.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_14310927_ayxnxkrpahml6atf6fm7mfgvnczdv5bv.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_14311495_refaunj64cbf4rf3zobifluyio0hzmkh.jpg yuvj444p 300 400    DecodeFAIL
decode_jpeg/400_f_14311552_tmefk12yayjvorxyzmh9j92623glx7zh.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_14311635_tvumkyrtgl2lnyouhfqp7urpxo1o8kxr.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/400_f_14312182_t8o4hv9r9eqs17mc1zzykfaxeufdh0p0.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_14313041_senx94ixyp2iokaqxsmas6aq43pzmfue.jpg yuvj444p 400 288    DecodeFAIL
decode_jpeg/400_f_14313562_yh6xdckpetsqd6cpzymfmjoy66uwboqz.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_14316145_utxi4cegtqh0m5iobzbwyns5osumck8w.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_5643155_upbvgu5cooibzcw4g5f4hgglijyumdqw.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/400_f_6674077_w0c4zo0owuaqwdgd37aeztu437c2felb.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_7095312_qdpqkhefixx37xfjtkly03lgnly2d3mt.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/400_f_7717462_0kkv7ej7lfp1ztqtbhjzx419t8goqkeh.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/415107223_273915da90.jpg yuvj444p 500 400    DecodeFAIL
decode_jpeg/439263359_dff7ea8381.jpg yuvj444p 500 400    DecodeFAIL
decode_jpeg/4point5.jpg yuvj444p 688 600    DecodeFAIL
decode_jpeg/4_tile.jpg yuvj444p 400 400    DecodeFAIL
decode_jpeg/548-3262vishnumurthi-teyyam-ritual-nellikkathuruthi-temple-malabar-coast-kerala-state-india-posters.jpg yuvj444p 400 300    DecodeFAIL
decode_jpeg/7sisters.bmp_RGB_444_100.jpg yuvj444p 1024 692    DecodeFAIL
decode_jpeg/a17_jb00519.jpg yuvj444p 4288 2848    DecodeFAIL
decode_jpeg/annual-1-tumb.jpg yuvj444p 500 332    DecodeFAIL
decode_jpeg/arbonne.jpg yuvj444p 300 528    DecodeFAIL
decode_jpeg/canona20_010421-1106-44.jpg yuvj422p 1600 1200    DecodeFAIL
decode_jpeg/canona520_img_0259.jpg yuvj422p 2272 1704    DecodeFAIL
decode_jpeg/canona95_img_8468.jpg yuvj422p 2592 1944    DecodeFAIL
decode_jpeg/canondigitalixus300_010319-0950-20.jpg yuvj422p 1600 1200    DecodeFAIL
decode_jpeg/canonditalixus330_020403-1144-46.jpg yuvj422p 1600 1200    DecodeFAIL
decode_jpeg/canoneos1ds_021104-1112-01-capt1.jpg yuvj444p 4072 2712    DecodeFAIL
decode_jpeg/canoneos5dmarkii_img_0001-acr.jpg yuvj444p 5616 3744    DecodeFAIL
decode_jpeg/capture28.jpg yuvj420p 904 4340    DecodeFAIL
decode_jpeg/capture30.jpg yuvj420p 860 1616    DecodeFAIL
decode_jpeg/capture31.jpg yuvj420p 982 1522    DecodeFAIL
decode_jpeg/capture37.JPG yuvj420p 1280 968    DecodeFAIL
decode_jpeg/capture39.JPG yuvj420p 1280 968    DecodeFAIL
decode_jpeg/capture40.JPG yuvj420p 1280 968    DecodeFAIL
decode_jpeg/capture47.jpg yuvj420p 860 1534    DecodeFAIL
decode_jpeg/cave.jpg yuvj420p 1600 1200    DecodeFAIL
decode_jpeg/collage1.JPG yuvj420p 1280 968    DecodeFAIL

Tianhaol, 2018-01-16:
Debugging into this function CodechalDecodeJpeg::DecodeStateLevel,  I found the format of output surface is NV12 and chromaType is jpegYUV444, error is returned.
     //Three new formats from HSW C0,HSW ULT can only be supported in specific conditions.
    if (jpegPicState.dwOutputFormat == Format_NV12 ||
        jpegPicState.dwOutputFormat == Format_YUY2 ||
        jpegPicState.dwOutputFormat == Format_UYVY)
    {
        //Only interleaved single scan are supported.
        if (m_jpegPicParams->m_totalScans != 1 ||
            m_jpegPicParams->m_interleavedData == 0)
        {
            return MOS_STATUS_UNKNOWN;
        }

        switch (m_jpegPicParams->m_chromaType)
        {
        case jpegYUV420:
        case jpegYUV422H2Y:
        case jpegYUV422H4Y:
            break;
        case jpegYUV422V2Y:
        case jpegYUV422V4Y:
            if (GFX_IS_GEN_8_OR_LATER(m_hwInterface->GetPlatform()) &&
                jpegPicState.dwOutputFormat == Format_NV12)
            {
                break;
            }
        default:
            return MOS_STATUS_UNKNOWN;
        }

XinfengZhang, 2018-06-12:
two root causes
1.  is in the gstreamer: https://bugzilla.gnome.org/show_bug.cgi?id=796505 
2.  middle ware should create 444 surface and image for the 444 jpeg src. but seems, it create 420 image and surface. it's incorrect behavior.

wangzj0601, 2018-07-12:
This issue can not be duplicated with the latest media-driver commit 4a85bb1, so close it

closed 2018-07-12 by wangzj0601