intel/media-driver #44: [gstreamer-vaapi-master][iHD_SKL] [encode] "inner_rawvideoparse: Internal data stream error" error message occurs during run encoding cases for AVC/MPEG2/HEVC/JPEG with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. take one case each test suite for example, run command
Avc encode
gst-launch-1.0 filesrc location=/media/encoderbitstreams//720p5994_parkrun_ter.yuv ! videoparse format=i420 width=1280 height=720 framerate=30 ! vaapih264enc rate-control=cqp init-qp=26 keyframe-period=1 num-slices=1 max-bframes=0 quality-level=1 ! video/x-h264,profile=main ! h264parse ! filesink location=./720p5994_parkrun_i420_profilemain_rc0_i_qp26_slice1_gop30_tu1_fps30.h264

Hevc encode
gst-launch-1.0 filesrc location=/media/encoderbitstreams//4K_2013_3840_2160_300frames.yuv ! videoparse format=i420 width=3840 height=2160 framerate=25 ! vaapih265enc rate-control=cqp init-qp=26 keyframe-period=30 num-slices=4 max-bframes=2 quality-level=4 ! video/x-265,profile=main ! h265parse ! filesink location=./4K_3840x2160_i420_profilemain_rc0_ipb_qp26_slice4_gop30_tu4_fps25.h265

Mpeg2 encode
gst-launch-1.0 filesrc location=/media/encoderbitstreams//720p5994_parkrun_ter.yuv ! videoparse format=i420 width=1280 height=720 framerate=30 ! vaapimpeg2enc rate-control=cqp quantizer=26 keyframe-period=1 max-bframes=0 quality-level=1 ! mpegvideoparse ! mpegtsmux ! filesink location=./720p5994_parkrun_i420_profilemain_rc0_i_qp26_slice1_gop29_tu1_fps30.mpg

Jpeg encode
gst-launch-1.0 filesrc location=/media/encoderbitstreams//MOBILE_352x288_30_orig_01.yuv '!' videoparse format=i420 width=704 height=576 '!' vaapijpegenc quality=20 '!' filesink location=MOBILE_352x288_30_orig_01.yuv.jpeg

3. error info. as below
0:00:00.021428060 21441      0x17eac00 ERROR                  vaapi gstvaapiencoder.c:435:gst_vaapi_encoder_put_frame: failed to encode frame (status = -1)
0:00:00.021458995 21441      0x17eac00 ERROR            vaapiencode gstvaapiencode.c:722:gst_vaapiencode_handle_frame: failed to encode frame 0 (status -1)
ERROR: from element /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse: Internal data stream error

XinfengZhang, 2017-12-26:
no vaapi level log. so cant locate the issue easily. need to reproduce it. are there detail log?

Tianhaol, 2018-01-16:
YV12 surface can be created in function DdiMedia_CreateSurfaces2,
but this format cannot be convert to OS format in DdiMedia_MediaSurfaceToMosResource.

Tianhaol, 2018-01-30:
If there is only one attribute with type VAConfigAttribRTFormat, failed to create config by driver.

wangzj0601, 2018-07-12:
This issue can not be duplicated with the latest media-driver commit 4a85bb18ecc9a18b7cf592d405c8c2fa1c7aad06, so close it

closed 2018-07-12 by wangzj0601