intel/media-driver #45: [gstreamer-vaapi-master][iHD_SKL] [VP8_encode] Feature vp8_encode can not be enabled during compile gst-vaapi with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. compile gst-vaapi component with command "./autogen.sh --disable-glx --prefix=/opt/yami/gstreamer10/ --enable-encoders" successfully
3. export GST_VAAPI_ALL_DRIVERS=1 due to vpg driver not in whitelist
4. run command "gst-inspect-1.0 vaapi", finding that vp8 encode does not be enabled

gst-inspect-1.0 vaapi
Plugin Details:
  Name                     vaapi
  Description              VA-API based elements
  Filename                 /opt/yami/gstreamer10/lib/gstreamer-1.0/libgstvaapi.so
  Version                  1.13.0.1
  License                  LGPL
  Source module            gstreamer-vaapi
  Source release date      2017-12-13 18:48 (UTC)
  Binary package           gstreamer-vaapi
  Origin URL               http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer

  vaapijpegdec: VA-API JPEG decoder
  vaapimpeg2dec: VA-API MPEG2 decoder
  vaapih264dec: VA-API H264 decoder
  vaapivc1dec: VA-API VC1 decoder
  vaapivp8dec: VA-API VP8 decoder
  vaapih265dec: VA-API H265 decoder
  vaapipostproc: VA-API video postprocessing
  vaapidecodebin: VA-API Decode Bin
  vaapisink: VA-API sink
  vaapimpeg2enc: VA-API MPEG-2 encoder
  vaapih265enc: VA-API H265 encoder
  vaapijpegenc: VA-API JPEG encoder
  vaapih264enc: VA-API H264 encoder
  vaapih264feienc: VA-API H264 FEI Advanced encoder (Experimental)

  14 features:
  +-- 14 elements

XinfengZhang, 2017-12-26:
#24

XinfengZhang, 2018-04-03:
fixed by #113 and #86

XinfengZhang, 2018-04-12:
no VP8 encode support , please try APL or CNL

closed 2018-04-12 by XinfengZhang