intel/media-driver #46: [gstreamer-vaapi-master][iHD_SKL]"vaapisink0: Internal error: could not render surface" error message occurs when run command using option "vaapisink" to show on screen with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. take one case for example, run command
gst-launch-1.0 filesrc location=/media/h264/freh10.264 ! h264parse ! vaapih264dec ! vaapipostproc width=336 ! vaapipostproc skin-tone-enhancement=TRUE ! vaapisink
3. error info. as below
** (gst-launch-1.0:21742): WARNING **: gstvaapidisplay.c:990: invalid property id 3 for "hue" of type 'GParamFloat' in 'GstVaapiDisplayX11'

** (gst-launch-1.0:21742): WARNING **: gstvaapidisplay.c:990: invalid property id 4 for "saturation" of type 'GParamFloat' in 'GstVaapiDisplayX11'

** (gst-launch-1.0:21742): WARNING **: gstvaapidisplay.c:990: invalid property id 5 for "brightness" of type 'GParamFloat' in 'GstVaapiDisplayX11'

** (gst-launch-1.0:21742): WARNING **: gstvaapidisplay.c:990: invalid property id 6 for "contrast" of type 'GParamFloat' in 'GstVaapiDisplayX11'
ERROR: from element /GstPipeline:pipeline0/GstVaapiSink:vaapisink0: Internal error: could not render surface

XinfengZhang, 2017-12-26:
it should related with DisplayAttribute, we discuss it in #17

XinfengZhang, 2018-04-09:
#fixed by #131

closed 2018-04-09 by XinfengZhang