intel/media-driver #47: [gstreamer-vaapi-master][iHD_SKL] [VPP] "tsdemux0: Internal data stream error" error message occurs during run vpp transcoding with iHD driver on SKL

wangzj0601, 2017-12-20:
Test Env:
============================================
Platform:               SKL
Arch:          		x86_64
Kernel:			4.12.0-rc2
libva:			master branch df544cd5a31e54d4cbd33a391795a8747ddaf789 
                        repo: https://github.com/01org/libva.git
media_driver:		master branch 81796c8a9e44b878e26064c898e0f4730e8220cf
                        repo: https://github.com/intel/media-driver
gmmlib:                 repo: https://github.com/intel/gmmlib
gst-vaapi:		master branch b0d41c5db8c29b3df2a5b23d3da55141d701b992

Reproduce Steps:
============================================
1. build env as above lists or according to wiki https://github.com/intel/media-driver
2. take one case for example, run command
gst-launch-1.0 filesrc location=/media/ts/Sally.ts ! tsdemux ! mpegvideoparse  ! vaapimpeg2dec ! vaapih264enc rate-control=cqp init-qp=4 keyframe-period=1 max-bframes=0 cabac=true  ! qtmux faststart=true '!' filesink location=test.mp4
3. error info. as below
ERROR: from element /GstPipeline:pipeline0/GstTSDemux:tsdemux0: Internal data stream error.
Additional debug info:
mpegtsbase.c(1613): mpegts_base_loop (): /GstPipeline:pipeline0/GstTSDemux:tsdemux0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll

FurongZhang, 2017-12-20:
Hi,

From the log, it failed in Demux. 

ERROR: from element /GstPipeline:pipeline0/GstTSDemux:tsdemux0: Internal data stream error.
 Additional debug info:
 mpegtsbase.c(1613): mpegts_base_loop (): /GstPipeline:pipeline0/GstTSDemux:tsdemux0:
 streaming stopped, reason not-negotiated (-4)

Thanks,
Furong

feiwan1, 2018-04-09:
@FurongZhang , without demux, issue also can be duplicated by using cmd:
"LIBVA_TRACE=./vaapi-driver gst-launch-1.0 -v videotestsrc num-buffers=100 ! video/x-raw,width=1280,height=720,format=NV12 ! vaapih264enc rate-control=cqp init-qp=4 keyframe-period=1 max-bframes=0 cabac=true ! qtmux faststart=true '!' filesink location=test.mp4"

Same cmdline can be run on vaapi driver, but fail on media driver. So I believe media driver have some compatibility issue with gstreamer-vaapi.

And I attach va trace when run the cmd line on vaapi driver and media driver, hope this can help you to debug.
[media-driver.143635.thd-0x00001aaa.log](https://github.com/intel/media-driver/files/1888271/media-driver.143635.thd-0x00001aaa.log)
[media-driver.143635.thd-0x00001aab.log](https://github.com/intel/media-driver/files/1888272/media-driver.143635.thd-0x00001aab.log)
[vaapi-driver.143849.thd-0x00001adf.log](https://github.com/intel/media-driver/files/1888273/vaapi-driver.143849.thd-0x00001adf.log)
[vaapi-driver.143849.thd-0x00001ae0.log](https://github.com/intel/media-driver/files/1888274/vaapi-driver.143849.thd-0x00001ae0.log)
[vaapi-driver.143849.thd-0x00001ae2.log](https://github.com/intel/media-driver/files/1888275/vaapi-driver.143849.thd-0x00001ae2.log)

FurongZhang, 2018-04-12:
Thanks for the log. Will check the log and get back to you soon.

Jister, 2018-05-02:
@feiwan1 With the latest gstreamer-vaapi, this case can be executed successfully, so it should be a bug of gstreamer-vaapi and has been fixed by the latest version.

Jister, 2018-05-02:
@wangzj0601 I have tried to debug on your case, it failed in tsdemux, before that, the media driver hasn't been called. So I also agree with Furong's opinion, the failure may be caused by tsdemux. 
And I'll keep my attention on this issue.

wangzj0601, 2018-05-03:
This issue can not be duplicated with gstreamer-vaapi 1.15.0.1 now, commit 2d95089a8d35a4a8ff5b04ec08a151f43196c2b4, so close it.

closed 2018-05-03 by wangzj0601