intel/media-driver #52: [DDI] MediaLibvaCaps constructor should be defined as protected

shural, 2017-12-26:
Given factory pattern is used to instantiate MediaLibvaCaps, the constructor should be accessible to public. Or should check null pointer of m_mediaCtx in the constructor.

yakuizhao, 2017-12-26:
Thanks for reporting the issue.
Since the MediaLibvaCaps is only used as the base class,  it looks reasonable that the constructor in MediaLibvaCaps is declared as protected. In such case it won't be called explicitly.

For the check of m_mediaCtx also makes sense. Of course as the m_mediaCtx is already checked in some other places. I think that this check is optional. Without it, it will hurt anything.

XinfengZhang, 2018-05-08:
fixed by d2d9f4e316b1fe4cc275a7dd0dee5e47a7783a70

closed 2018-05-08 by XinfengZhang