intel/media-driver #54: [DDI] VAEntrypointStats should not be bounded to AVC FEI

shural, 2017-12-26:
VAEntrypointStats is generic, and can be used for both AVC and HEVC FEI, and may later for other encoders. It's not good coding behavior to bound VAEntrypointStats to AVC FEI in MediaLibvaCaps::GetEncodeCodecMode.

Also, if in regular calling sequence that DDI decode or encode key is already parsed from input parameters such as profile and vaentrypoint, suggest use the key to map to CODECHAL keys defined in lower layer (i.e., CODECHAL) instead of parsing the same profile and vaentrypoint again.

XinfengZhang, 2017-12-26:
changing it to be bounded with AVC will also impact other api. such as VAStatus vaQueryConfigEntrypoints, for the given profile such as VAProfileH264Main, it should report VAEntrypointStats .

kelvinhu325, 2017-12-26:
Yes, VAEntrypointStats is generic, but driver implements VAEntrypointStats with the same path as AVC currently

XinfengZhang, 2019-08-27:
close the issue, it will be changed after HEVC preEnc is enable. the Entrypoint could be combined for different codec

closed 2019-08-27 by XinfengZhang