intel/media-driver #55: [DDI] MediaLibvaCaps::CreateVPConfig pass in unused parameters

shural, 2017-12-26:
It's hard to understand the purpose of unused input parameters. Is it better to just change the function signature to remove those unused parameters? Or if VP intends to use those in the future, at least make a comment in function definition.

FurongZhang, 2018-06-19:
@shural , thanks a lot for your feedback. We will make a comment in function definition.

FurongZhang, 2018-07-16:
@shural We clarified the issue about unused parameters in function in https://github.com/intel/media-driver/commit/bbb88087e5244a020b963dd3ce5e6a22a7b1fed2. Please review it and let us know if you have any concern about that within one week. We will close this issue if no further feedback in 2 weeks. Thank you in advance!

closed 2018-07-31 by XinfengZhang