intel/media-driver #57: [SKL][VP8 DEC] VP8 DEC lead to GPU reset in iHD.

mypopydev, 2017-12-27:
FFmpeg VA-API Vp8 DEC work with iHD driver will lead to GPU reset, and if skip the first Byte in macroblock_offset, this issue is disappear. I give a workround in FFmpeg, more information you can refer to the link: [https://github.com/mypopydev/FFmpeg/commit/8f78dbea03056d125f3afe2148b5b658ec537eee](https://github.com/mypopydev/FFmpeg/commit/8f78dbea03056d125f3afe2148b5b658ec537eee)

XinfengZhang, 2017-12-27:
macroblock offset should be calculated base on whether it's align with 8bits

lakulako, 2018-08-09:
Try instead of #239:
https://github.com/lakulako/media-driver/pull/1
https://github.com/Intel-Media-SDK/MediaSDK/pull/493
macroblock_offset should be exact number of consumed bits after range in bool decoder made >=128.

closed 2018-11-12 by XinfengZhang