intel/media-driver #58: [SKL][VC1 DEC] Need to skip emulation prevention byte to calc macroblock offset in VC1 DEC

mypopydev, 2017-12-27:
Need to skip emulation prevention byte to calc macroblock offset in VC1 DEC, but as VA-API, emulation prevention byte will be handle in driver, now iHD handle emulation prevention byte in H.264/H.265 DEC, but didn't handle in VC1 DEC.

And we have a workround ([https://github.com/mypopydev/FFmpeg/commit/8f78dbea03056d125f3afe2148b5b658ec537eee](https://github.com/mypopydev/FFmpeg/commit/8f78dbea03056d125f3afe2148b5b658ec537eee)) for this, but I think need to fix in iHD driver.

yakuizhao, 2017-12-29:
Thanks for reporting the issue.
Yes.  This can be handled in driver like H264/H265.

But as one middleware tries to adjust the macroblock for VC1,  we need to let them know the update. Otherwise it will be broken after it is added into the driver.

XinfengZhang, 2018-01-19:
@mypopydev seems your WA is for VP8. it is not related with the description.

closed 2018-11-12 by XinfengZhang