intel/media-driver #63: Fail to build cmrt if libva is installed at custom location

dvrogozh, 2017-12-29:
Try to build media stack as follows:

```
export CUSTOM_DIR=/path/to/custom/location  # not /usr or /usr/inlcude
cd libva
./autogen.sh --prefix=$CUSTOM_DIR
make && make install

export PKG_CONFIG_PATH=$CUSTOM_DIR/lib/pkgconfig
mkdir -p build && cd build
cmake ../media-driver \
  -DMEDIA_VERSION="2.0.0" \
  -DBUILD_ALONG_WITH_CMRTLIB=1 \
  -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ \
  -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ \
  -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ \
  -DBS_DIR_MEDIA=`pwd`/../media-driver \
  -DCMAKE_INSTALL_PREFIX=$CUSTOM_DIR \
  -DCMAKE_INSTALL_LIBDIR=$CUSTOM_DIR/lib \
  -DINSTALL_DRIVER_SYSCONF=OFF \
  -DLIBVA_DRIVERS_PATH=$CUSTOM_DIR/dri
make
```
You will see the following error:
Scanning dependencies of target igfxcmrt
```
make[2]: Leaving directory `/data/dvrogozh/git/github/build'
make[2]: Entering directory `/data/dvrogozh/git/github/build'
[  0%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_printf_host.cpp.o
[  1%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_rt_helpers.cpp.o
[  1%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_avs_state_msg_ex.cpp.o
[  1%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/hardware/cm_device.cpp.o
[  1%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/hardware/cm_device_export.cpp.o
[  2%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/hardware/cm_surface_manager.cpp.o
[  2%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/hardware/cm_timer.cpp.o
[  2%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/hardware/cm_queue.cpp.o
[  2%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/hardware/cm_perf_statistics.cpp.o
[  2%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/hardware/cm_device_os.cpp.o
[  3%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/hardware/cm_surface_manager_os.cpp.o
[  3%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/hardware/cm_device_export_os.cpp.o
[  3%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/hardware/cm_performance.cpp.o
[  3%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/hardware/cm_timer_os.cpp.o
[  4%] Linking CXX shared library igfxcmrt64.so
/usr/bin/ld: cannot find -lva
collect2: error: ld returned 1 exit status
make[2]: *** [cmrtlib/linux/igfxcmrt64.so] Error 1
make[2]: Leaving directory `/data/dvrogozh/git/github/build'
make[1]: *** [cmrtlib/linux/CMakeFiles/igfxcmrt.dir/all] Error 2
make[1]: Leaving directory `/data/dvrogozh/git/github/build'
make: *** [all] Error 2
make: Leaving directory `/data/dvrogozh/git/github/build'
```

uartie, 2018-01-09:
Your `-DLIBVA_DRIVERS_PATH=$CUSTOM_DIR/dri` should be `-DLIBVA_DRIVERS_PATH=$CUSTOM_DIR/lib/dri`

Nonetheless, both "compile" configuration settings seem to work for me with custom libva installation path (although, probably won't work at runtime with former since libva will default search in `$CUSTOM_DIR/lib/dri`)

uartie, 2018-01-09:
Can you share your output for your `cmake ../media-driver ...` command?

PR #64 (2018-01-17): Fix cmrt build if libva installed at custom location
This fix also removes cmrt dependency from libva-x11 which it actually
does not have.

Fixes #63

Change-Id: I85ea2f5b2f330bc1bae77f15e2a32b9a8e4adf9d
commit 5c37e4fc (2018-01-17): Fix cmrt build if libva installed at custom location

This fix also removes cmrt dependency from libva-x11 which it actually
does not have.

Fixes #63

Change-Id: I85ea2f5b2f330bc1bae77f15e2a32b9a8e4adf9d
closed 2018-01-17 by oliver-sang