intel/media-driver #70: Compilation fails with missing defines

ViGill, 2018-01-08:
It looks like the typdefs such as FALSE, TRUE, PVOID,.. are not accessible from the files that require it.
For instance:

In file included from /workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/os/mos_os.h:34:0,
                 from /workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_common.h:35,
                 from /workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/codec/hal/codechal.h:29,
                 from /workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.h:31,
                 from /workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_downsampling.cpp:27:
/workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_downsampling.cpp: In member function 'bool FieldScalingInterface::IsFieldScalingSupported(CODECHAL_DECODE_PROCESSING_PARAMS*)':
/workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/os/mos_defs.h:271:78: error: 'TRUE' was not declared in this scope
 #define MOS_WITHIN_RANGE(_x, _min, _max)  (((_x >= _min) && (_x <= _max)) ? (TRUE) : (FALSE))
                                                                              ^
/workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_downsampling.cpp:523:10: note: in expansion of macro 'MOS_WITHIN_RANGE'
     if (!MOS_WITHIN_RANGE(srcSurface->dwWidth, m_minInputWidth, m_maxInputWidth)     ||
          ^
/workspace/dev/qsv/git/media-driver/media_driver/agnostic/common/os/mos_defs.h:271:87: error: 'FALSE' was not declared in this scope
 #define MOS_WITHIN_RANGE(_x, _min, _max)  (((_x >= _min) && (_x <= _max)) ? (TRUE) : (FALSE))
                                                                                       ^

Did i miss something ?

Thanks

yakuizhao, 2018-01-09:
@ViGill Thanks for reporting the issue.
Not sure whether you follow the build instr that is mentioned in https://github.com/intel/media-driver?
Will you please help to list which commit id is used for media-driver and GmmLib?
Thanks

ViGill, 2018-01-09:
Hi,

I use master (0154744f9c2d18d8a09a570eff90a452d155720a) for GmmLib and the recommended afc3466ce8c28873dcad31d1a166b5a5977add5f for media-driver.
I follow the readme line by line yes.

I have a lot of missing typedef/defines... (PVOID, PFLOAT, TRUE,...)

oliver-sang, 2018-01-10:
sorry, I think the afc3466ce8c28873dcad31d1a166b5a5977add5f is also a commit in GmmLib repo.
We don't have a recommended commit for media for now. Since you are using the tip of GmmLib now (that 0154744f9c2d18d8a09a570eff90a452d155720a), I suggest you to use the tip of media-driver, too. Or at 4bff0eba01d78896dafa7682d60b55d3cd8f839b which is compatible with the latest GmmLib.

yakuizhao, 2018-01-10:
@ViGill , What @oliver-sang mentioned is correct.
Now the GmmLib does some cleanup and removes the mentioned type definition.
And the driver code should also be updated. Otherwise the old driver code can't be built against on the latest GmmLib as the required type is removed in GmmLib.

Thanks

ViGill, 2018-01-10:
Understood thanks, I will test

oliver-sang, 2018-01-12:
@ViGill could we close this?

ViGill, 2018-01-14:
Hi, will tell you soon :)

ViGill, 2018-01-17:
Hi, with gmmlib b1451bbe4c506f17ddc819f12b4b448fc08698c5 and latest master it compiles fine. Thanks

closed 2018-01-17 by ViGill