intel/media-driver #74: Some Unit Test give a Error message, but test result is OK

mypopydev, 2018-01-12:
I build media-driver with : 
cmake ../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_VERSION="2.0.0" -DBUILD_ALONG_WITH_CMRTLIB=1 -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -DBUILD_TYPE=Debug

get some suspicious unit test log and result, this is sample as follow:

[----------] 5 tests from Surface2DTest
[ RUN      ] Surface2DTest.MultipleSizes
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
[... 86 lines cut ...]
[CM]:   CRITICAL - Surface2DSanityCheck:122: Error: Invalid surface height.
[CM]:   CRITICAL - CreateSurface2D:1006: Error: Surface 2D sanity check failure.
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
[CM]:   CRITICAL - Surface2DSanityCheck:122: Error: Invalid surface height.
[CM]:   CRITICAL - CreateSurface2D:1006: Error: Surface 2D sanity check failure.
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
[CM]:   CRITICAL - Surface2DSanityCheck:122: Error: Invalid surface height.
[CM]:   CRITICAL - CreateSurface2D:1006: Error: Surface 2D sanity check failure.
Load driver from ../../../iHD_drv_video.so.
INFO: Load driver success
INFO: Found init function __vaDriverInit_1_0 and CM entry point vaCmExtSendReqMsg
.INFO: Get driver DDI functions
drmIoctl: with unsupport IOType
[CM]:   CRITICAL - Surface2DSanityCheck:122: Error: Invalid surface height.
[CM]:   CRITICAL - CreateSurface2D:1006: Error: Surface 2D sanity check failure.
[       OK ] Surface2DTest.MultipleSizes (133 ms)
[ RUN      ] Surface2DTest.ReadWrit

shengcon, 2018-01-12:
These error messages are expected. Some cases are designed to pass wrong parameters to CM driver. CM driver captures wrong input, prints a message, and returns an error code. And the case completes subsequently.

XinfengZhang, 2018-01-18:
@mypopydev

mypopydev, 2018-01-18:
@XinfengZhang  you can close this RFC, thanks.

oliver-sang, 2018-01-29:
Close according to @mypopydev feedback

closed 2018-01-29 by oliver-sang