intel/media-driver #83: [Request] Clean the assert message

mypopydev, 2018-01-19:
now https://github.com/intel/media-driver/blob/e7f3a83d509997d77cf046adb4e8600326525100/media_driver/linux/common/os/mos_context_specific.cpp#L449 give a lot of assert message in debug mode, can you guys help to clean the assert message.

BTW: I've grep the "I915_CONTEXT_PARAM_SSEU" in libdrm/linux kernel, but can't find any  information for this, only get a patch in https://patchwork.kernel.org/patch/9707859/ but didn't merge to linux kernel.

LuoIvan, 2018-01-19:
Not a issue. I915_CONTEXT_PARAM_SSEU is a new extension, the corresponding kmd patch will up-streamed soon.

mypopydev, 2018-01-19:
@LuoIvan Before this part upstream to kernel finish, how to handle this case? I think media-driver need to query kernel whether of not support I915_CONTEXT_PARAM_SSEU, not hard coding as this way.

LuoIvan, 2018-02-11:
Unfortunately, query the support to I915_CONTEXT_PARAM_SSEU also require kmd change. 
Currently there is no perfect way to solve this problem, considering the assert msg only impact the debug driver, suggest we can depriortize or close this issue now.

dvrogozh, 2018-02-13:
There is no way to query i915 driver for support of context parameter. However, we can cover non-upstream uAPIs usages with the macro and configure-stage option which will be switch off by default assuming the default usage to be open source one. Those who want to try the extension, will apply i915 change and rebuild UMD with the enabled config option.

closed 2019-08-27 by XinfengZhang