intel/media-driver #88: GCC 6.x+ compilation errors (-Werror=deprecated-declarations)

uartie, 2018-01-23:
Compilation fails with GCC 6.x+ since,
```
commit fa0d40ee6d3d7ec34478da166b2740eaf9b7c0ce
Author: xuan_(oliver)__sang <oliver.sang@intel.com>
Date:   Sun Jan 21 17:38:35 2018 +0800

    treat warning as error for Linux build(CL#732090)
    
    Change-Id: I24c93e7d2cf2d800d9ad67a82dd2c4adf698c1fc
```

The compilation errors are:
```
[  9%] Building CXX object media_driver/CMakeFiles/iHD_drv_video.dir/linux/common/codec/ddi/media_ddi_encode_avc.cpp.o
/home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp: In member function ‘CODEC_AVC_PROFILE_IDC DdiEncodeAvc::GetAVCProfileFromVAProfile()’:
/home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp:141:10: error: ‘VAProfileH264Baseline’ is deprecated [-Werror=deprecated-declarations]
     case VAProfileH264Baseline:
          ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/ddi/media_libva.h:30:0,
                 from /home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp:27:
/home/uaeoff/Work/workspace/media/install/include/va/va.h:359:5: note: declared here
     VAProfileH264Baseline va_deprecated_enum = 5,
     ^~~~~~~~~~~~~~~~~~~~~
/home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp: In member function ‘VAStatus DdiEncodeAvc::ParsePackedHeaderParams(void*)’:
/home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp:1655:47: error: ‘VAEncPackedHeaderH264_SEI’ is deprecated [-Werror=deprecated-declarations]
     else if (encPackedHeaderParamBuf->type == VAEncPackedHeaderH264_SEI)
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/uaeoff/Work/workspace/media/install/include/va/va.h:4181:0,
                 from /home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/ddi/media_libva.h:30,
                 from /home/uaeoff/Work/workspace/media/build/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp:27:
/home/uaeoff/Work/workspace/media/install/include/va/va_enc_h264.h:125:5: note: declared here
     VAEncPackedHeaderH264_SEI va_deprecated_enum  = (0x80000000 | 1),
     ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video.dir/build.make:7743: media_driver/CMakeFiles/iHD_drv_video.dir/linux/common/codec/ddi/media_ddi_encode_avc.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:164: media_driver/CMakeFiles/iHD_drv_video.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

The deprecations are in libva and have been deprecated since libva 2.0.

GCC 7.x is the default compiler on Ubuntu 17.10 (artful) and Fedora 26 and 27.  GCC 6.x is default on Ubuntu 17.04 (zesty).

oliver-sang, 2018-01-24:
thanks for reporting this. we will look at this.
For now, you could revert change in fa0d40ee6d3d7ec34478da166b2740eaf9b7c0ce to unblock the build on higher version gcc. Sorry for inconvenience.

uartie, 2018-02-07:
Would be nice if we could turn off `-Werror` with a cmake configure flag... especially since different compiler versions often generate different warnings that aren't always addressed (or tested) immediately in driver sourcecode.

oliver-sang, 2018-02-07:
a good idea!
How about a flag MEDIA_TREAT_WARNING_AS_ERROR? default will be 'yes'. in case there are some new warning show up in whatever compiler, could explicitly use -DMEDIA_TREAT_WARNING_AS_ERROR='no' to make compilation pass before having a final fix.

uartie, 2018-02-07:
Yes, this would work.  How about `MEDIA_WARNING_AS_ERROR` instead?

closed 2018-04-03 by oliver-sang