intel/media-driver #92: fails when using Media SDK HEVC FEI Encoding Sample with CmDevice LoadProgramISA error

DeadBread, 2018-01-29:
I am trying to launch Media SDK HEVC FEI Encoding Sample and get the output like this:
```
Intel(R) Media SDK HEVC FEI Encoding Sample Version 8.2.25.

Pipeline         :	Encode
Resolution       :	1920x1088
Frame rate       :	30/1 = 30.00
Processing order :	Display
Gop Size         :	1
GopOptFlag       :	Open, Not Strict
IDR Interval     :	Infinite
GopRefDist       :	1
NumRefFrame      :	4
NumRefActiveP    :	3
NumRefActiveBL0  :	3
NumRefActiveBL1  :	1
B-pyramid        :	Off
P-pyramid        :	SIMPLE

Media SDK version	1.25
Error in loading ISA file.CmDevice LoadProgramISA error
```
I've tried to debug through this and found out, that apparently methode 

> CMRTKernelBase::LoadProgramISA

is trying to load file by this path 

> /opt/intel/mediasdk/lib64/hme_downscale_genx.isa

and can't find it for it is not there. 
And I could not locate this file anywhere on my computer.

XinfengZhang, 2018-01-30:
HEVC FEI is using CMRT , and will load isa files. we will fix it.

XinfengZhang, 2018-03-08:
fixed by https://github.com/intel/media-driver/commit/f58de0e1b6cf1b52771e5c54b179e9a9cf6751b7

closed 2018-03-08 by XinfengZhang