intel/media-driver #93: Segmentation fault (core dumped) when build iHD driver

mypopydev, 2018-01-30:
media@s0n0c0:~/Souces/umd_dev/media-driver$ git branch -v
* master a03392e [Codec] add the missing cache setting for VDBOX.(CL#734336)

media@s0n0c0:~/Souces/umd_dev/build$ uname -a
Linux s0n0c0 4.13.0-21-generic #24-Ubuntu SMP Mon Dec 18 17:29:16 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

media@s0n0c0:~/Souces/umd_dev/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

cmake ../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_VERSION="2.0.0" -DBUILD_ALONG_WITH_CMRTLIB=1 -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -DBUILD_TYPE=Debug

media@s0n0c0:~/Souces/umd_dev/build$ make clean && make -j `nproc`

Then get a Segmentation fault (core dumped) like this:

Running drvult...
[==========] Running 34 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 1 test from MediaCapsDdiTest
[ RUN      ] MediaCapsDdiTest.DecodeEncodeProfile
[       OK ] MediaCapsDdiTest.DecodeEncodeProfile (10 ms)
[----------] 1 test from MediaCapsDdiTest (10 ms total)

[----------] 2 tests from MediaDecodeDdiTest
[ RUN      ] MediaDecodeDdiTest.DecodeHEVCLong
[       OK ] MediaDecodeDdiTest.DecodeHEVCLong (6 ms)
[ RUN      ] MediaDecodeDdiTest.DecodeAVCLong
[       OK ] MediaDecodeDdiTest.DecodeAVCLong (8 ms)
[----------] 2 tests from MediaDecodeDdiTest (14 ms total)

[----------] 2 tests from MediaEncodeDdiTest
[ RUN      ] MediaEncodeDdiTest.EncodeHEVC_DualPipe
Segmentation fault (core dumped)
media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/build.make:62: recipe for target 'RunULT' failed
make[2]: *** [RunULT] Error 139
CMakeFiles/Makefile2:299: recipe for target 'media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all' failed
make[1]: *** [media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

oliver-sang, 2018-01-30:
Sorry for inconvenience.
This is ULT crash and specific for debug type build, could you first disable ULT running by passing
-DBYPASS_MEDIA_ULT=yes
in cmake command line if you want to build debug version driver?

We will fix this problem.

(BTW, you also could build release (-DBUILD_TYPE=release) or release-internal (-DBUILD_TYPE=release-internal) version)

XinfengZhang, 2018-02-07:
fixed by https://github.com/intel/media-driver/commit/a703ec7362fedede71940929a1a18d9577324173

mypopydev, 2018-02-08:
Tested and verified, LGTM

mypopydev, 2018-02-26:
LGTM,  you can close this issue. @oliver-sang @XinfengZhang

closed 2018-02-28 by XinfengZhang