intel/media-driver #97: What's the difference from https://github.com/intel/intel-vaapi-driver

YijinLiu, 2018-02-01:
I am using https://github.com/intel/intel-vaapi-driver now. I wonder what's the advantages this new driver provides. Should I switch to it?

YijinLiu, 2018-02-06:
We are mainly using vaapi to decode / encode h264/vp8 on Goldmont/Skylake. Is there any performance difference between the two drivers? Like cpu/gpu/memory usage or video quality/compression ratio.

mypopydev, 2018-02-09:
@YijinLiu I think now media-driver can't support VP8 encode in Skylake, so I guess maybe i965 is the only  choice in this case.

XinfengZhang, 2018-02-11:
Initial releases of the Intel® Media Driver for VAAPI  will track close with Intel vaapi media driver features, quality and performance.  It is expected that future release with focus from Intel and the community will deliver best in class features, quality and performance.

enihcam, 2018-07-14:
@XinfengZhang 
My notebook is KabyLake based, and in the most of time it is just used for decoding video, so in performance/power-consumption aspects which driver should I use?

Additionally, in wiki can you create a table sheet to list the mappings between processors and drivers?

francoism90, 2018-07-31:
Any update on this? Does one need both on supported platforms?

XinfengZhang, 2018-08-02:
no need both, you can just choose either for current platform, but in future, from CNL+, only media_driver exist. no intel-vaapi-driver for CNL+ platform.

enihcam, 2018-08-02:
@XinfengZhang I exported chromium-vaapi GPU reports for both drivers, and found that intel-media-driver supports VP9 hardware acceleration, but intel-vaapi-driver does not. My processor is i5-7200U.

closed 2018-09-21 by YijinLiu

sandsmark, 2019-12-23:
here on whiskey lake both intel-vaapi-driver and intel-media-driver reports to support vp9 via vainfo, and intel-vaapi-driver reports decoding support for two more formats (VAProfileH264StereoHigh and VAProfileH264MultiviewHigh).

Testing with a VP9 test video from https://test-videos.co.uk/bigbuckbunny/webm-vp9 with mpv I get the same CPU usage with both intel-vaapi-driver and intel-media-driver as well.

Am I missing something? The reason I started looking at this is because I get this seemingly harmless error now; `[ffmpeg] AVHWDeviceContext: Failed to query surface attributes: 20 (the requested function is not implemented).`