intel/media-driver #98: SegFault: DdiEncode_CreateContext --> GmmLib::GmmResourceInfoCommon

uartie, 2018-02-02:
Since https://github.com/intel/gmmlib/commit/d6c3b56745798bdd0c7e4920ebb5b4192225c17b, segmentation fault occurs during vaCreateContext in test_va_api CreateContext conformance tests:

Command: `LIBVA_DRIVER_NAME=iHD test_va_api --gtest_filter=*CreateContext*`
GDB Backtrace: [gdb.txt](https://github.com/intel/media-driver/files/1688013/gdb.txt)

Possibly related to https://github.com/intel/gmmlib/issues/6

oliver-sang, 2018-02-02:
Thanks for reporting this issue. We've observed this problem and added a wiki about we should use https://github.com/intel/gmmlib/commit/6f565367e87c2c6628efd8d60b357b5ffb1a54e8 for now.( [PR#99](https://github.com/intel/media-driver/pull/99) also mentioned this )

We are under debug to solve this issue. Sorry for any inconvenience.

uartie, 2018-02-02:
#35 request would prevent this.

uartie, 2018-02-02:
Keep in mind that I'm reporting the issue in the scope of our CI failures... it does not make sense for us to revert commits in CI just to make things happy.

oliver-sang, 2018-02-02:
thanks, we'll seriously consider the proposal about git submodule.

uartie, 2018-02-02:
Thanks.  The only other option is when things break/regress is to fix them immediately ;)

XinfengZhang, 2018-02-07:
fixed by https://github.com/intel/media-driver/commit/a703ec7362fedede71940929a1a18d9577324173

mypopydev, 2018-02-08:
Tested and verified, LGTM

closed 2018-04-03 by oliver-sang