intel/media-driver #105: libva error: /usr/local/lib/dri/iHD_drv_video.so init failed

foxdavion, 2018-02-14:
Hello! Help me((
When I try to initialize the driver, I get an error, what could be the problem?

`root@ubuntu:/opt/intel/mediasdk/samples# ./sample_multi_transcode -hw -i::h264 /opt/msdk/tests/content/test_stream.mpeg2 -o::h264 out.h264
Multi Transcoding Sample Version 8.2.25.

libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 2
libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 2
error: failed to initialize VAAPI device

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), main, transcode.Init failed at /opt/msdk/samples/sample_multi_transcode/src/sample_multi_transcode.cpp:844
`
`root@ubuntu:/opt/intel# vainfo
error: can't connect to X server!
libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 2
vaInitialize failed with error code 2 (resource allocation failed),exit
`
`root@ubuntu:/opt/intel/mediasdk/samples# uname -a
Linux ubuntu 4.4.0-87-generic #110-Ubuntu SMP Tue Jul 18 12:55:35 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
`
`root@ubuntu:/opt/intel/mediasdk/samples# lspci -nn -s 0:02.0
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:5912] (rev 04)
`
`root@ubuntu:/opt/intel/mediasdk/samples# lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 158
Model name:            Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
Stepping:              9
CPU MHz:               799.968
CPU max MHz:           4500.0000
CPU min MHz:           800.0000
BogoMIPS:              8400.75
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              8192K
NUMA node0 CPU(s):     0-7
`

foxdavion, 2018-02-14:
root@ubuntu:/opt/intel/mediasdk/samples# lsmod | grep 915
parport                49152  2 ppdev,parport_pc
ib_iser                49152  0
rdma_cm                49152  1 ib_iser
ib_mad                 49152  2 ib_cm,ib_sa
raid10                 49152  0
i915_bpo             1306624  1
intel_ips              20480  1 i915_bpo
i2c_algo_bit           16384  1 i915_bpo
drm_kms_helper        155648  1 i915_bpo
drm                   364544  3 i915_bpo,drm_kms_helper
video                  40960  1 i915_bpo
root@ubuntu:/opt/intel/mediasdk/samples# sudo lspci -k | grep -EA2 'VGA|3D'
00:02.0 VGA compatible controller: Intel Corporation Device 5912 (rev 04)
        DeviceName:  Onboard IGD
        Subsystem: Micro-Star International Co., Ltd. [MSI] Device 7991

XinfengZhang, 2018-02-22:
@foxdavion sorry for late response. it's CNY.  could you tell us the driver version and libva version? especially the commit number 
thanks

foxdavion, 2018-02-25:
Hi!
root@ubuntu:/opt/libva# git rev-parse HEAD
f2f9956f1dd871af59f2fd9bc6fdfc6c0992f0d1

root@ubuntu:/opt/media/media-driver# git rev-parse HEAD
43a84e7a893ebf0433a8efc6bb7ab6a3e325a3fa

root@ubuntu:/opt/media/gmmlib# git rev-parse HEAD
6f565367e87c2c6628efd8d60b357b5ffb1a54e8

root@ubuntu:/opt/media/build/media_driver/linux/ult/ult_app# ./drvult
[==========] Running 34 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from MediaEncodeDdiTest
[ RUN      ] MediaEncodeDdiTest.EncodeHEVC_DualPipe
DRM_IOCTL_I915_GEM_APERTURE failed: Inappropriate ioctl for device
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [25]
param: 4, val: 0
/opt/media/media-driver/media_driver/linux/ult/ult_app/ddi_test_encode.cpp:60: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 1
Platform = 0, Failed function = driverLoader.InitDriver

foxdavion, 2018-02-25:
Ubuntu 16.04.04 install:

apt-get install build-essential
cd /opt
apt install autoconf libtool libdrm-dev xorg xorg-dev openbox libx11-dev libgl1-mesa-glx libgl1-mesa-dev
git clone https://github.com/01org/libva
cd libva/
git reset --hard f2f9956f1dd871af59f2fd9bc6fdfc6c0992f0d1
./autogen.sh
make
make install

git clone https://github.com/intel/media-driver
git clone https://github.com/intel/gmmlib
cd gmmlib
git reset --hard 6f565367e87c2c6628efd8d60b357b5ffb1a54e8
<workspace>
    |- gmmlib
    |- media-driver
mkdir build
 ./unified_cmake.sh
 make -j8
make install
git clone https://github.com/Intel-Media-SDK/MediaSDK msdk
cd msdk/
perl tools/builder/build_mfx.pl --cmake=intel64.make.release
make -j8 -C __cmake/intel64.make.release
make install

root@ubuntu:/opt/intel/mediasdk/samples#  ./sample_multi_transcode -hw -i::h264 /opt/msdk/tests/content/test_stream.mpeg2 -o::h264 out.h264
Multi Transcoding Sample Version 8.2.25.

libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 2
libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 2
error: failed to initialize VAAPI device

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), main, transcode.Init failed at /opt/msdk/samples/sample_multi_transcode/src/sample_multi_transcode.cpp:844

:(((

oliver-sang, 2018-02-26:
could you first check whether the iHD_drv_video.so is really installed under /usr/local/lib/dri?
after install, do you have a /etc/profile.d/intel-media.sh? And if so, what's its content?
Thanks

foxdavion, 2018-02-26:
root@ubuntu:/etc/profile.d# cat /etc/profile.d/intel-media.sh
# add libva driver path/name exporting for intel media solution
export LIBVA_DRIVERS_PATH=/usr/local/lib/dri
export LIBVA_DRIVER_NAME=iHD

root@ubuntu:/usr/local/lib/dri# ls -l
total 78660
-rw-r--r-- 1 root root 80545760 Feb 25 19:08 iHD_drv_video.so

root@ubuntu:/usr/local/lib# ls -l
total 856
drwxr-xr-x 2 root root    4096 Feb 25 19:09 dri
-rwxr-xr-x 1 root root     971 Feb 25 18:52 libva-drm.la
lrwxrwxrwx 1 root root      18 Feb 25 18:52 libva-drm.so -> libva-drm.so.2.0.0
lrwxrwxrwx 1 root root      18 Feb 25 18:52 libva-drm.so.2 -> libva-drm.so.2.0.0
-rwxr-xr-x 1 root root   52792 Feb 25 18:52 libva-drm.so.2.0.0
-rwxr-xr-x 1 root root    1077 Feb 25 18:52 libva-glx.la
lrwxrwxrwx 1 root root      18 Feb 25 18:52 libva-glx.so -> libva-glx.so.2.0.0
lrwxrwxrwx 1 root root      18 Feb 25 18:52 libva-glx.so.2 -> libva-glx.so.2.0.0
-rwxr-xr-x 1 root root   89008 Feb 25 18:52 libva-glx.so.2.0.0
-rwxr-xr-x 1 root root     917 Feb 25 18:52 libva.la
lrwxrwxrwx 1 root root      14 Feb 25 18:52 libva.so -> libva.so.2.0.0
lrwxrwxrwx 1 root root      14 Feb 25 18:52 libva.so.2 -> libva.so.2.0.0
-rwxr-xr-x 1 root root  517640 Feb 25 18:52 libva.so.2.0.0
-rwxr-xr-x 1 root root     993 Feb 25 18:52 libva-x11.la
lrwxrwxrwx 1 root root      18 Feb 25 18:52 libva-x11.so -> libva-x11.so.2.0.0
lrwxrwxrwx 1 root root      18 Feb 25 18:52 libva-x11.so.2 -> libva-x11.so.2.0.0
-rwxr-xr-x 1 root root  178976 Feb 25 18:52 libva-x11.so.2.0.0
drwxr-xr-x 2 root root    4096 Feb 25 18:52 pkgconfig
drwxrwsr-x 4 root staff   4096 Feb 25 18:45 python2.7
drwxrwsr-x 3 root staff   4096 Feb 25 17:39 python3.5

XinfengZhang, 2018-02-26:
from log, the driver is opened , but failed to call __vaDriverInit_1_0
seems it is related with previous GMM related change

XinfengZhang, 2018-02-26:
main engineer effort of media-driver is focus on CNL+ platform. of course, we also support BDW&SKL, and is considering whether there are real requirement for other platform. could you send a mail to me to discuss platform support requirement? carl.zhang@intel.com

foxdavion, 2018-02-26:
Yes, kabylake.

foxdavion, 2018-02-26:
its works?!!!!!
```
root@ubuntu:/opt/media/build/media_driver/linux/ult/ult_app# LD_PRELOAD=../libdrm_mock/libdrm_mock.so ./drvult ../../../iHD_drv_video.so
[==========] Running 34 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from MediaDecodeDdiTest
[ RUN      ] MediaDecodeDdiTest.DecodeHEVCLong
[       OK ] MediaDecodeDdiTest.DecodeHEVCLong (3 ms)
[ RUN      ] MediaDecodeDdiTest.DecodeAVCLong
[       OK ] MediaDecodeDdiTest.DecodeAVCLong (2 ms)
[----------] 2 tests from MediaDecodeDdiTest (5 ms total)

[----------] 1 test from MediaCapsDdiTest
[ RUN      ] MediaCapsDdiTest.DecodeEncodeProfile
[       OK ] MediaCapsDdiTest.DecodeEncodeProfile (1 ms)
[----------] 1 test from MediaCapsDdiTest (1 ms total)

[----------] 2 tests from MediaEncodeDdiTest
[ RUN      ] MediaEncodeDdiTest.EncodeHEVC_DualPipe
[       OK ] MediaEncodeDdiTest.EncodeHEVC_DualPipe (6 ms)
[ RUN      ] MediaEncodeDdiTest.EncodeAVC_DualPipe
[       OK ] MediaEncodeDdiTest.EncodeAVC_DualPipe (6 ms)
[----------] 2 tests from MediaEncodeDdiTest (12 ms total)

[----------] 4 tests from KernelTest
[ RUN      ] KernelTest.LoadDestroyProgram
[       OK ] KernelTest.LoadDestroyProgram (31 ms)
[ RUN      ] KernelTest.LoadWrongIsa
[       OK ] KernelTest.LoadWrongIsa (7 ms)
[ RUN      ] KernelTest.CreateKernel
[       OK ] KernelTest.CreateKernel (12 ms)
[ RUN      ] KernelTest.SetArgument
[       OK ] KernelTest.SetArgument (38 ms)
[----------] 4 tests from KernelTest (88 ms total)

[----------] 1 test from TaskTest
[ RUN      ] TaskTest.CreateDestroy
[       OK ] TaskTest.CreateDestroy (6 ms)
[----------] 1 test from TaskTest (6 ms total)

[----------] 3 tests from Surface2DUPTest
[ RUN      ] Surface2DUPTest.MultipleSizes
[       OK ] Surface2DUPTest.MultipleSizes (62 ms)
[ RUN      ] Surface2DUPTest.InvalidSystemMemory
[       OK ] Surface2DUPTest.InvalidSystemMemory (12 ms)
[ RUN      ] Surface2DUPTest.PlanarFormat
[       OK ] Surface2DUPTest.PlanarFormat (25 ms)
[----------] 3 tests from Surface2DUPTest (99 ms total)

[----------] 7 tests from Surface2DTest
[ RUN      ] Surface2DTest.MultipleSizes
[       OK ] Surface2DTest.MultipleSizes (74 ms)
[ RUN      ] Surface2DTest.PlanarFormats
[       OK ] Surface2DTest.PlanarFormats (30 ms)
[ RUN      ] Surface2DTest.DestroyNull
[       OK ] Surface2DTest.DestroyNull (5 ms)
[ RUN      ] Surface2DTest.ReadWrite
[       OK ] Surface2DTest.ReadWrite (10 ms)
[ RUN      ] Surface2DTest.ReadWriteWithStride
[       OK ] Surface2DTest.ReadWriteWithStride (5 ms)
[ RUN      ] Surface2DTest.Initialization
[       OK ] Surface2DTest.Initialization (6 ms)
[ RUN      ] Surface2DTest.GetInfo
[       OK ] Surface2DTest.GetInfo (10 ms)
[----------] 7 tests from Surface2DTest (140 ms total)

[----------] 3 tests from ThreadSpaceTest
[ RUN      ] ThreadSpaceTest.MultipleSizes
[       OK ] ThreadSpaceTest.MultipleSizes (10 ms)
[ RUN      ] ThreadSpaceTest.DependencyPattern
[       OK ] ThreadSpaceTest.DependencyPattern (10 ms)
[ RUN      ] ThreadSpaceTest.MediaWalkingPattern
[       OK ] ThreadSpaceTest.MediaWalkingPattern (10 ms)
[----------] 3 tests from ThreadSpaceTest (30 ms total)

[----------] 3 tests from BufferTest
[ RUN      ] BufferTest.MultipleSizes
[       OK ] BufferTest.MultipleSizes (33 ms)
[ RUN      ] BufferTest.ReadWrite
[       OK ] BufferTest.ReadWrite (5 ms)
[ RUN      ] BufferTest.Initialization
[       OK ] BufferTest.Initialization (5 ms)
[----------] 3 tests from BufferTest (43 ms total)

[----------] 2 tests from BufferUPTest
[ RUN      ] BufferUPTest.MultipleSizes
[       OK ] BufferUPTest.MultipleSizes (31 ms)
[ RUN      ] BufferUPTest.InvalidSystemMemory
[       OK ] BufferUPTest.InvalidSystemMemory (10 ms)
[----------] 2 tests from BufferUPTest (41 ms total)

[----------] 4 tests from Surface3DTest
[ RUN      ] Surface3DTest.MultipleSizes
[       OK ] Surface3DTest.MultipleSizes (51 ms)
[ RUN      ] Surface3DTest.Abgr16Format
[       OK ] Surface3DTest.Abgr16Format (10 ms)
[ RUN      ] Surface3DTest.ReadWrite
[       OK ] Surface3DTest.ReadWrite (5 ms)
[ RUN      ] Surface3DTest.Initialization
[       OK ] Surface3DTest.Initialization (5 ms)
[----------] 4 tests from Surface3DTest (71 ms total)

[----------] 2 tests from QueueTest
[ RUN      ] QueueTest.CreateTwice
[       OK ] QueueTest.CreateTwice (6 ms)
[ RUN      ] QueueTest.EnqueueWithoutTask
[       OK ] QueueTest.EnqueueWithoutTask (5 ms)
[----------] 2 tests from QueueTest (11 ms total)

[----------] Global test environment tear-down
[==========] 34 tests from 12 test cases ran. (548 ms total)
[  PASSED  ] 34 tests.

```

XinfengZhang, 2018-02-27:
ULT test driver functions and has no dependency with HW. till now media-driver does not support KabyLake. thanks

closed 2018-02-27 by foxdavion

Brainiarc7, 2018-03-30:
I have the exact problem but on Apollo Lake, specifically the Intel(R) Atom(TM) Processor E3950.

avago-lsi-starcore, 2018-04-27:
foxdavion, what exactly fixed your problem?

HumanFred, 2018-04-27:
foxdavion, what exactly fixed your problem?

enihcam, 2018-05-17:
@foxdavion Do you have a plan/timeline for media-driver supporting Kabylake?

DanitiKain, 2021-01-08:
Did this ever get resolved for Kabylake?

dvrogozh, 2021-01-08:
Media driver supports KBL and APL. See list of supported platforms in readme: https://github.com/intel/media-driver#supported-platforms. Repeating here (as of today's):
- BDW (Broadwell)
- SKL (Skylake)
- BXT (Broxton) / APL (Apollo Lake)
- KBLx (KBL/Kaby Lake; CFL/Coffe Lake; WHL/Whiskey Lake; CML/Comet Lake; AML/Amber Lake)
- ICL (Ice Lake)
- JSL (Jasper Lake)/EHL (Elkhart Lake)
- TGL (Tiger Lake)
- DG1/SG1

In particular, driver recognizes those platforms which were mentioned in this issue (KBL 5912 and APL 5a84):
* https://github.com/intel/media-driver/blob/fcdacb40e128dfeaa95d6c9166d547dea72221ac/media_driver/linux/gen9/ddi/media_sysinfo_g9.cpp#L805
* https://github.com/intel/media-driver/blob/fcdacb40e128dfeaa95d6c9166d547dea72221ac/media_driver/linux/gen9/ddi/media_sysinfo_g9.cpp#L763

I believe Ubuntu 20.04 has media driver package version 20.1.1 which supports KBL and APL mentioned in this issue.

jengajenga, 2021-01-09:
> its works?!!!!!
>
> ```
> root@ubuntu:/opt/media/build/media_driver/linux/ult/ult_app# LD_PRELOAD=../libdrm_mock/libdrm_mock.so ./drvult ../../../iHD_drv_video.so
> [==========] Running 34 tests from 12 [...]

very selfish of you to close this without telling others how you got it to work yet many chipped in to help you.