intel/media-driver #108: How to disable testing during build?

pauldotknopf, 2018-03-02:
Is there a CMAKE property I can set?

pauldotknopf, 2018-03-02:
The reason is because a bunch of the CMake test stuff is manually overwriting the CMAKE* flags required for cross compiling.

For example: https://github.com/intel/media-driver/blob/79fbea02d08cf86430f178e224e293f0d3ba50cb/media_driver/linux/ult/ult_app/CMakeLists.txt#L55

If the scripts aren't going to be cross-compile friendly, you should have a way to (CMAKE variable) to disable tests and apps.

oliver-sang, 2018-03-07:
sorry for late. There is no such a CMAKE flag could use now.
For a temporary solution, you could remove https://github.com/intel/media-driver/blob/0cf7f92eaac821a289461192fbaa0e6415b7c6db/media_driver/media_top_cmake.cmake#L23

Thanks for pointing out such usage scenario, we will add one flag to disable the ult

DavidMarec, 2018-03-19:
In addition to remove `CMAKE_{lang}_FLAGS` overwrites, please avoid to hard define the compiler;
as it has been previously set in the toolchain file.

```cmake
set(CMAKE_C_COMPILER "g++") 
```
vs
```cmake
set(CMAKE_C_COMPILER "${CMAKE_CXX_COMPILER}") 
```

But at the end, it will be more convenient to rename _C++_ file  extensions.

oliver-sang, 2018-03-20:
thanks for suggestion! we will look into this.

oliver-sang, 2018-06-15:
there are actually two options could be used now -
MEDIA_RUN_TEST_SUITE (ON/OFF) OFF will disable the build of ult
BYPASS_MEDIA_ULT - if this is defined and equal to 'yes', the ult won't run even when it build

closed 2018-06-15 by oliver-sang