intel/media-driver #110: avc vdenc doesn't work as expected when using FFmpeg.

xhaihao, 2018-03-08:
The slice header in the encoded h264 stream is wrong, ffprobe reports the following errors: 

```
[h264 @ 0x559068a82fa0] illegal modification_of_pic_nums_idc 32
[h264 @ 0x559068a82fa0] decode_slice_header error
[h264 @ 0x559068a82fa0] no frame!
[h264 @ 0x559068a82fa0] illegal modification_of_pic_nums_idc 32
[h264 @ 0x559068a82fa0] decode_slice_header error
[h264 @ 0x559068a82fa0] no frame!
[h264 @ 0x559068a81c30] decoding for stream 0 failed
```

The command line for encoding is ' ffmpeg -loglevel debug -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -s:v WxH -i <input yuv> -vf 'format=nv12,hwupload' -low_power 1 -c:v h264_vaapi output.h264'

ffmpeg commit: 00ae5c1d3d87efc7f13b86cce80e139c805db179
media-driver: ef454928dd0df8b3f4492be921ed87cfad0324d4
gmmlib: b8d74ac3fcea0dffa901ce62399d43a937212074
libva: b8d74ac3fcea0dffa901ce62399d43a937212074

XinfengZhang, 2018-03-13:
reproduced, will fix it

XinfengZhang, 2018-03-28:
correct one comments, it's is a CNL only issue ---- not verified. maybe also exist with SKL

closed 2018-11-12 by XinfengZhang