intel/media-driver #111: Invalid jpeg structure in jpeg file encoded by FFmpeg/mjpeg_vaapi

xhaihao, 2018-03-09:
Command:  ffmpeg -loglevel debug -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v WxH -i input.yuv -vf 'format=nv12,hwupload' -frames 1  -c:v mjpeg_vaapi output.jpeg

However the encoded file has invalid jpeg file structure

> $> display output.jpeg
> display-im6.q16: Invalid JPEG file structure: two SOI markers `output.jpeg' @ error/jpeg.c/JPEGErrorHandler/322.
> 

ffmpeg commit: 00ae5c1d3d87efc7f13b86cce80e139c805db179
media-driver: ef45492
gmmlib: b8d74ac3fcea0dffa901ce62399d43a937212074
libva: 417b0184d4a767721e9b0a5f1597907707f545cf

XinfengZhang, 2018-03-13:
reproduced , will fix it

Tianhaol, 2018-06-13:
Disable APP data in media driver, JPEG file can be encoded, but there are corruptions when opened.
The input src surface dumped by codechal is different the input yuv file.

mypopydev, 2018-06-14:
I think the correct command is:  ffmpeg -loglevel debug -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -f rawvideo -pix_fmt nv12  -s:v WxH -i input.yuv -vf 'format=nv12,hwupload' -frames 1 -c:v mjpeg_vaapi output.jpeg

xhaihao, 2018-06-14:
In my test, input.yuv is a yuv420p file, and ffmpeg will convert it to NV12 due to "-vf 'format=nv12,hwupload'"

PR #273 (2018-08-20): Jpeg encode full packed header support 
1. check whether the raw data include all of the headers, and provide different solution
2. when the packed data include all of the headers, scale the qmatrix for quality setting.
fix #111 #232
closed 2018-08-20 by Sherry-Lin