intel/media-driver #112: The driver fails to encode H.265 videos on CNL when using FFmpeg/hevc_vaapi

xhaihao, 2018-03-09:
> [hevc_vaapi @ 0x562dec275dc0] Param buffer (24) is 0x4.
> [CODEC]:CRITICAL - EncodeBrcFrameUpdateKernel:5609: MOS returned error, eStatus = 0x2
> [CODEC]:CRITICAL - EncodeKernelFunctions:7070: MOS returned error, eStatus = 0x2
> [CODEC]:CRITICAL - ExecuteKernelFunctions:3005: MOS returned error, eStatus = 0x2
> [CODEC]:CRITICAL - ExecuteEnc:4277: ENC failed.
> [CODEC]:CRITICAL - Execute:555: MOS returned error, eStatus = 0x2
> [LIBVA]:CRITICAL - EncodeInCodecHal:362: DDI:Failed in Codechal!
> [LIBVA]:CRITICAL - EndPicture:80: DDI:DdiEncode_EncodeInCodecHal return failure.
> [hevc_vaapi @ 0x562dec275dc0] Failed to end picture encode issue: 24 (internal encoding error).
> [hevc_vaapi @ 0x562dec275dc0] Encode failed: -5.
> Video encoding failed

The command: ffmpeg -loglevel debug -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v WxH  -i input.yuv -vf 'format=nv12,hwupload' -c:v hevc_vaapi output.h265

ffmpeg commit: 00ae5c1d3d87efc7f13b86cce80e139c805db179
media-driver: ef45492
gmmlib: b8d74ac3fcea0dffa901ce62399d43a937212074
libva: 417b0184d4a767721e9b0a5f1597907707f545cf

The same command works well on SKL with this driver.

XinfengZhang, 2018-03-13:
reproduced , will fix it

XinfengZhang, 2018-03-15:
root caused, current media-driver just support I B B1 B2 frames, not support P frame on CNL

closed 2019-08-27 by XinfengZhang