intel/media-driver #114: [gstreamer-vaapi][avc/mpeg2/h265] doesn't support 1080P encode.

feiwan1, 2018-03-16:
System Environment
=======
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver 48f33e511e08a9335d57aad4c09acbbb4aeb0a7f
gst-vaapi tag 1.13.90

Reproduce Steps
==============
$LIBVA_TRACE=./avcenc gst-launch-1.0 videotestsrc num-buffers=100 ! video/x-raw,format=NV12,width=1920,height=1080 ! vaapih264enc ! fakesink

feiwan1, 2018-03-16:
[avcenc.123919.thd-0x00001509.log](https://github.com/intel/media-driver/files/1819200/avcenc.123919.thd-0x00001509.log)
[avcenc.123919.thd-0x00001508.log](https://github.com/intel/media-driver/files/1819201/avcenc.123919.thd-0x00001508.log)

uartie, 2018-03-16:
@feiwan1 can you share how to whitelist iHD driver in gst-vaapi?  I tried:
```
$ LIBVA_DRIVER_NAME=iHD gst-launch-1.0 videotestsrc num-buffers=100 ! video/x-raw,format=NV12,width=1920,height=1080 ! vaapih264enc ! fakesink
WARNING: erroneous pipeline: no element "vaapih264enc"
```

feiwan1, 2018-03-18:
@uartie , try to export GST_VAAPI_ALL_DRIVERS=1.

jcescamilla, 2018-03-30:
Locally verified this using fix #122 and it works for avc and h265.
mpeg2 is currently blocked by issue  #116

KseniyaTikhomirova, 2018-05-31:
@jcescamilla Hi, are you going to disable this check for Gen10+ platforms? vaCreateContext on CNL for HEVC encoding of HD and fullHD streams returns VA_STATUS_ERROR_RESOLUTION_NOT_SUPPORTED because of the same check in media_libva_caps_g10.cpp

dmitryermilov, 2018-05-31:
+1  to Kseniya's question

jcescamilla, 2018-06-01:
Hi @KseniyaTikhomirova and @dmitryermilov ,
Yes, we can disable this check for Gen10. I'll work on a fix and give an update asap.

jcescamilla, 2018-06-06:
@KseniyaTikhomirova and @dmitryermilov 
I've submitted a pull-request: https://github.com/intel/media-driver/pull/179
Please feel free to add any feedback.

XinfengZhang, 2018-07-31:
@feiwan1 could we close the issue?

closed 2018-11-12 by XinfengZhang