intel/media-driver #115: [gstreamer-vaapi][h265enc] h265enc doesn't work.

feiwan1, 2018-03-18:
System Environment
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver 48f33e5
gst-vaapi tag 1.13.90

Reproduce Steps
$LIBVA_TRACE=./h265enc gst-launch-1.0 videotestsrc num-buffers=100 ! vaapih265enc ! fakesink

feiwan1, 2018-03-18:
[h265enc.074241.thd-0x00003e53.log](https://github.com/intel/media-driver/files/1822463/h265enc.074241.thd-0x00003e53.log)
[h265enc.074241.thd-0x00003e54.log](https://github.com/intel/media-driver/files/1822464/h265enc.074241.thd-0x00003e54.log)

jcescamilla, 2018-03-29:
This caused because Gstreamer tries checking for ROI QP delta support flag. Added a PR that fixes this. #132

closed 2018-04-03 by XinfengZhang