intel/media-driver #116: [gstreamer-vaapi][mpeg2enc] mpeg2 encode doesn't work.

feiwan1, 2018-03-18:
System Environment
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver 48f33e5
gst-vaapi tag 1.13.90

Reproduce Steps
$ LIBVA_TRACE=./mpeg2enc gst-launch-1.0 videotestsrc num-buffers=100 ! vaapimpeg2enc ! fakesink

feiwan1, 2018-03-18:
[mpeg2enc.081058.thd-0x00004338.log](https://github.com/intel/media-driver/files/1822482/mpeg2enc.081058.thd-0x00004338.log)
[mpeg2enc.081058.thd-0x00004335.log](https://github.com/intel/media-driver/files/1822483/mpeg2enc.081058.thd-0x00004335.log)
[mpeg2enc.081058.thd-0x00004334.log](https://github.com/intel/media-driver/files/1822484/mpeg2enc.081058.thd-0x00004334.log)

jcescamilla, 2018-03-30:
media-driver does not support VAEncPackedHeaderSequence. Will propose to check for this on gstreamer and not use if driver does not supported.

feiwan1, 2019-06-20:
Junyan already set a patch to gst-vaapi to fixed this issue:
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/merge_requests/105

If there is no need behavior in media-driver side, I am going to close this. Thanks.

feiwan1, 2019-06-26:
Fixed in gst-vaapi side:
https://cgit.freedesktop.org/gstreamer/gstreamer-vaapi/commit/?id=de51eb00599232d3311e6bc42fc168b04b86a027
Patch merged, and close this issue.

Thanks

closed 2019-06-26 by feiwan1