intel/media-driver #118: [gstreamer-vaapi][jpegenc] jpeg encode doesn't work.

feiwan1, 2018-03-18:
System Environment
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver 48f33e5
gst-vaapi tag 1.13.90

Reproduce Steps
LIBVA_TRACE=./jpegenc gst-launch-1.0 videotestsrc num-buffers=1 ! video/x-raw,format=NV12 ! vaapijpegenc ! filesink location=a.jpeg

check a.jpeg, all green mosaic.

feiwan1, 2018-03-18:
[jpegenc.090800.thd-0x00004a85.log](https://github.com/intel/media-driver/files/1822537/jpegenc.090800.thd-0x00004a85.log)
[jpegenc.090800.thd-0x00004a86.log](https://github.com/intel/media-driver/files/1822538/jpegenc.090800.thd-0x00004a86.log)
[jpegenc.090800.thd-0x00004a89.log](https://github.com/intel/media-driver/files/1822539/jpegenc.090800.thd-0x00004a89.log)
[jpegenc.090800.thd-0x00004a8b.log](https://github.com/intel/media-driver/files/1822540/jpegenc.090800.thd-0x00004a8b.log)

jcescamilla, 2018-03-30:
Unable to reproduce this issue with the latest code.
libva commit [3921874](https://github.com/intel/libva/commit/392187428cee9f5062c1b989cbe83a0a5e17c861)
intel-media-driver [3d843ba](https://github.com/intel/media-driver/commit/3d843ba4d3f9fb31363a257eae3fa751bd818e3c)
gst-vaapi [966d4d0](https://github.com/GStreamer/gstreamer-vaapi/commit/966d4d0476a966438812ece0bef8540371c4974c)

XinfengZhang, 2018-04-03:
cant reproduce, close it firstly, @feiwan1 any concerns, please add comments

closed 2018-04-03 by XinfengZhang