intel/media-driver #119: [gstreamer-vaapi][vaapipostproc] vaapipostproc couldn't work with vaapisink.

feiwan1, 2018-03-19:
System Environment
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver 48f33e5
gst-vaapi tag 1.13.90

Reproduce Steps
root@yami-kbl02:~# LIBVA_TREAE=./postporc gst-launch-1.0 videotestsrc num-buffers=100  !  vaapipostproc format=23 ! vaapisink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapisink0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayX11\)\ vaapidisplayx11-1";
ERROR: from element /GstPipeline:pipeline0/GstVaapiSink:vaapisink0: Internal error: could not render surface
Additional debug info:
gstvaapisink.c(1475): gst_vaapisink_show_frame_unlocked (): /GstPipeline:pipeline0/GstVaapiSink:vaapisink0
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...

feiwan1, 2018-03-19:
[postporc.014536.thd-0x00005654.log](https://github.com/intel/media-driver/files/1823463/postporc.014536.thd-0x00005654.log)
[postporc.014536.thd-0x00005655.log](https://github.com/intel/media-driver/files/1823464/postporc.014536.thd-0x00005655.log)

XinfengZhang, 2018-03-28:
@FurongZhang  please take a look