intel/media-driver #124: [Build] cmake3 fails to find pthread.h for ULT

lakulako, 2018-03-22:
Command line
*cmake3 ../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_VERSION="2.0.0" -DBUILD_ALONG_WITH_CMRTLIB=1 -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -DCMAKE_INSTALL_LIBDIR=/usr/lib64 -DINSTALL_DRIVER_SYSCONF=OFF -DLIBVA_DRIVERS_PATH=/usr/lib64/dri -DBUILD_TYPE="debug" -DCMAKE_BUILD_TYPE=Debug*
gives:
*CMake Error at /usr/share/cmake3/Modules/CheckIncludeFile.cmake:56 (configure_file):
  configure_file attempted to configure a file:
  /.../media-driver/CMakeFiles/CMakeTmp/CheckIncludeFile.c into
  a source directory.
Call Stack (most recent call first):
  /usr/share/cmake3/Modules/FindThreads.cmake:139 (CHECK_INCLUDE_FILE)
  media_driver/linux/ult/ult_app/googletest/CMakeLists.txt:42 (find_package)
-- Configuring incomplete, errors occurred!*

pthread.h does exist. 
Helps commenting out in media_driver/media_top_cmake.cmake:
**#** ADD_SUBDIRECTORY(${CMAKE_CURRENT_LIST_DIR}/linux/ult)

oliver-sang, 2018-03-23:
I cannot reproduce this issue in my centos7.3. Both cmake 3.5 I installed and cmake3 worked fine.

$ cmake --version
cmake version 3.5.2

$ cmake3 --version
cmake3 version 3.6.3

I saw same results as below in both cases -
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE

Looked into your log, this part is most suspicious -
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
CMake Error at /usr/share/cmake3/Modules/CheckIncludeFile.cmake:56 (configure_file):
configure_file attempted to configure a file:
/.../media-driver/CMakeFiles/CMakeTmp/CheckIncludeFile.c into
a source directory.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

and this - media-driver/CMakeFiles - looks really like you source directory.

If looking into /usr/share/cmake3/Modules/CheckIncludeFile.cmake:56, it's as below -
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
    configure_file(${CMAKE_ROOT}/Modules/CheckIncludeFile.c.in
      ${CMAKE_BINARY_DIR}${CMAKE_FILES_DIRECTORY}/CMakeTmp/CheckIncludeFile.c)
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

So I double whether your CMAKE_BINARY_DIR was overridden by accident.

Could you please check whether you have CMakeFiles/ subfolder in your media-driver source tree? Is so, you have run cmake under source tree before and your source tree is kind of 'contaminated'. Then maybe the easiest way to check further is to clone a new clean source tree from github (git clone https://github.com/intel/media-driver.git) and retry.

lakulako, 2018-03-23:
Thanks a lot! Everything works fine after media-driver folder was cleaned.
Sorry for invalid issue.

closed 2018-03-23 by lakulako

oliver-sang, 2018-03-26:
not at all! Thanks for reporting issue and supplying detail information