intel/media-driver #125: [gmm] Incorrect assertion in debug mode

lakulako, 2018-03-22:
The driver is built with __debug__.
Command line, for example:
> sample_hevc_fei -encode -i input.yuv -o out.h265 -w 1920 -h 1080 -r 4 -g 12 -qp 30 -n 24  -EncodedOrder

during creating contexts assertion fails:
MEMORY_OBJECT_CONTROL_STATE GMM_STDCALL GmmLib::GmmCachePolicyCommon::CachePolicyGetMemoryObject(GMM_RESOURCE_INFO *pResInfo, GMM_RESOURCE_USAGE_TYPE Usage)
{
    const GMM_CACHE_POLICY_ELEMENT *CachePolicy = NULL;
    __GMM_ASSERT(pGmmGlobalContext->GetCachePolicyElement(Usage).Initialized);
...
Here Usage is 128, obtained from GmmResourceUsage[MosUsage=29]

Probably __GMM_ASSERT is defined only in debug and the code is obsolete, because commenting it out leads to normal execution.

oliver-sang, 2018-03-23:
sorry for inconvenience. We've known about this debug build issue. Currently, we are working on fix. Will update when we get progress.

oliver-sang, 2018-03-23:
@lakulako could you please try whether 1013df31b00b516097ac87da7599f16564e9690d fix your case? Thanks

lakulako, 2018-03-23:
It does help. Thank you!

closed 2018-03-26 by lakulako