intel/media-driver #127: Build error: libdrm_mock.so: undefined symbol: pci_system_cleanup

dbermond, 2018-03-26:
I'm getting this error when trying to compile Intel Media Driver:
```
[100%] Built target iHD_drv_video
Scanning dependencies of target RunULT
Running devult...
./devult: symbol lookup error: ../libdrm_mock/libdrm_mock.so: undefined symbol: pci_system_cleanup
make[2]: *** [media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/build.make:63: RunULT] Error 127
make[1]: *** [CMakeFiles/Makefile2:300: media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

Trying to track down the issue, I've found this:
- it builds fine when removing `-fno-plt` from the environment variables CFLAGS and CXXFLAGS (this flag is enabled by default in my distribution)
- it builds fine until commit 7792911ec7462fd9141caa2863b5d8a912f15350 when using `-fno-plt`
- it seems to be caused by commit 50fd991e14a8ee592695f53946896d9837efe9af. It fails to build from this commit onwards when using `-fno-plt`.

This breaks compilation in Arch Linux because this disbribution enables `-fno-plt` by default in CFLAGS and CXXFLAGS.

**System information:**
**OS:** Arch Linux x86_64
**Compiler:** gcc 7.3.1
**gmmlib:** git master
**libva:** git master
**libdrm:** 2.4.91

**Environment variables (system default):**
CPPFLAGS="-D_FORTIFY_SOURCE=2"
CFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt"
CXXFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
DEBUG_CFLAGS="-g -fvar-tracking-assignments"
DEBUG_CXXFLAGS="-g -fvar-tracking-assignments"

oliver-sang, 2018-03-27:
thanks for reporting! we're starting the investigation.

oliver-sang, 2018-04-03:
this should be fixed by d95d8f7ab7ac94a2e0f4ee6a4b4794898dc2d3b7. could you please check? Thanks