intel/media-driver #128: vaCreateImage can't support BGR0 format

mypopydev, 2018-03-28:
When use the cmd **"ffmpeg_g -f x11grab -video_size 2560x1440 -i :0 -f pulse -i default -acodec aac -vaapi_device /dev/dri/renderD128 -vsync 0 -movflags +faststart -vf hwupload,scale_vaapi=format=nv12 -c:v h264_vaapi 1522205981.mp4**"

get the error like this:

[AVHWFramesContext @ 0x5555579aabc0] Failed to create image for surface 0: 20 (the requested function is not implemented).
[hwupload @ 0x55555798b140] Failed to upload frame: -5.
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0

the root cause is  vaQueryImageFormats report support BGR0, but vaCreateImage create BGR0 format fail.

XinfengZhang, 2018-03-28:
will add it for vaCreateImage

XinfengZhang, 2018-03-30:
#133 fix the issue, @mypopydev  please review, to check whether it make your requirement

closed 2018-05-14 by XinfengZhang