intel/media-driver #129: in order to compile I had to revert 5700e289

ceyusa, 2018-03-28:
In order to compile current master (3d843ba) I got this error mesage:

```
/home/vjaquez/gst-va/master/media-driver/media_driver/agnostic/common/os/mos_os.c: In function ‘MEMORY_OBJECT_CONTROL_STATEMos_CachePolicyGetMemoryObject(MOS_HW_RESOURCE_DEF)’:
/home/vjaquez/gst-va/master/media-driver/media_driver/agnostic/common/os/mos_os.c:730:9: error: ‘CM_RESOURCE_USAGE_L1_Enabled_SurfaceState’ was not declared in this scope                                                                                                                                
         CM_RESOURCE_USAGE_L1_Enabled_SurfaceState, 
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/vjaquez/gst-va/master/media-driver/media_driver/agnostic/common/os/mos_os.c:730:9: note: suggested alternative: ‘MOS_CM_RESOURCE_USAGE_L1_Enabled_SurfaceState’
         CM_RESOURCE_USAGE_L1_Enabled_SurfaceState,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         MOS_CM_RESOURCE_USAGE_L1_Enabled_SurfaceState
media_driver/CMakeFiles/iHD_drv_video.dir/build.make:2774: recipe for target 'media_driver/CMakeFiles/iHD_drv_video.dir/agnostic/common/os/mos_os.c.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video.dir/agnostic/common/os/mos_os.c.o] Error 1
make[2]: *** Waiting for unfinished jobs.... 
```

It seems that CM_RESOURCE_USAGE_L1_Enabled_SurfaceState is not completely defined. By reverting commit 5700e289 the compilation ends.

ceyusa, 2018-03-28:
I guess I found the issue:

Since I followed the instruction to use gmmlib at commit 6f56536, the commit 6fb6598 in gmmlib is not included, thus CM_RESOURCE_USAGE_L1_Enabled_SurfaceState is not defined.

So, either the gmmlib commit restriction is obsolete, or commit 5700e289 should be reverted.

ceyusa, 2018-03-28:
I just tried master branch of gmmlib and media-driver compiles just fine. Then, I assume the compilation instructions are obsolete.

oliver-sang, 2018-03-29:
sorry for confusion. The wiki is out-of-date and I just updated. Yes, for now, the media-driver tip should work with gmmlib tip. If you think it's ok now, could you please close this issue. Sorry for any inconvenience.

ceyusa, 2018-03-29:
Perhaps a policy of using always master or specific releases on both projects would simplify a lot the tracking. 

Closing since the wiki for glibmm notes is now updated.

closed 2018-03-29 by ceyusa