intel/media-driver #134: [ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), main, transcode.Init failed

Brainiarc7, 2018-03-30:
Running the smoke tests after installing the iMSDK as illustrated here https://software.intel.com/en-us/articles/build-and-debug-open-source-media-stack results in the error above. 

I'm not logged on to the machine at the moment, but this gist explains how the toolkit was built: https://gist.github.com/Brainiarc7/df0b1c081a4ede3ea80d3f2bcbd3739a

I'll attach logs and sysinfo as soon as I'm logged on.

Brainiarc7, 2018-04-03:
Closing, issue resolved with a proper rebuild.

chaotaklon, 2018-04-21:
Hi, did you solve the problem? If yes, can you tell how to solve? Thanks.

Brainiarc7, 2018-04-21:
Here's how I built the package on Ubuntu:

**Satisfy baseline dependencies:**

`sudo apt-get -y install autoconf automake build-essential libass-dev libtool pkg-config texinfo zlib1g-dev libva-dev cmake mercurial libdrm-dev libvorbis-dev libogg-dev git libx11-dev libperl-dev libpciaccess-dev libpciaccess0 xorg-dev`

**Setup build environment:**

Work space init:

`mkdir -p ~/vaapi`

Build the dependency chain as shown:

1. Libva :

Libva is an implementation for VA-API (Video Acceleration API)

VA-API is an open-source library and API specification, which provides access to graphics hardware acceleration capabilities for video processing. It consists of a main library and driver-specific acceleration backends for each supported hardware vendor. It is a prerequisite for building the VAAPI driver components below.

```
cd ~/vaapi
git clone https://github.com/01org/libva
cd libva
./autogen.sh --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu
time make -j$(nproc) VERBOSE=1
sudo make -j$(nproc) install
sudo ldconfig -vvvv
```

2. Gmmlib:

The Intel(R) Graphics Memory Management Library provides device specific and buffer management for the Intel(R) Graphics Compute Runtime for OpenCL(TM) and the Intel(R) Media Driver for VAAPI.

The component is a prerequisite to the Intel Media driver build step below.

To build this, create a workspace directory within the vaapi sub directory and run the build:

```
mkdir -p ~/vaapi/workspace
cd ~/vaapi/workspace
git clone https://github.com/intel/gmmlib
mkdir -p build
cd build
cmake -DCMAKE_BUILD_TYPE= Release -DARCH= 64 ../gmmlib
make -j$(nproc)
```

3. Intel Media driver:

The Intel(R) Media Driver for VAAPI is a new VA-API (Video Acceleration API) user mode driver supporting hardware accelerated decoding, encoding, and video post processing for GEN based graphics hardware, released under the MIT license.

```
cd ~/vaapi/workspace
git clone https://github.com/intel/media-driver
cd media-driver
git submodule init
git pull
mkdir -p ~/vaapi/workspace/build_media
cd ~/vaapi/workspace/build_media

```
Configure the project with cmake:

```
cmake ../media-driver \
-DMEDIA_VERSION="2.0.0" \
-DBUILD_ALONG_WITH_CMRTLIB=1 \
-DBS_DIR_GMMLIB=$PWD/../gmmlib/Source/GmmLib/ \
-DBS_DIR_COMMON=$PWD/../gmmlib/Source/Common/ \
-DBS_DIR_INC=$PWD/../gmmlib/Source/inc/ \
-DBS_DIR_MEDIA=$PWD/../media-driver \
-DCMAKE_INSTALL_PREFIX=/usr \
-DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu \
-DINSTALL_DRIVER_SYSCONF=OFF \
-DLIBVA_DRIVERS_PATH=/usr/lib/x86_64-linux-gnu/dri
```

Then build the media driver:

`time make -j$(nproc) VERBOSE=1`

Then install the project:

`sudo make -j$(nproc) install VERBOSE=1`

Add yourself to the video group:

`sudo usermod -a -G video $USER`

Now, export environment variables as shown below:

```
LIBVA_DRIVERS_PATH=/usr/lib/x86_64-linux-gnu/dri
LIBVA_DRIVER_NAME=iHD
```

Put that in `/etc/environment`.

4. libva-utils:

This package provides a collection of tests for VA-API, such as vainfo, needed to validate a platform's supported features (encode, decode & postproc attributes on a per-codec basis by VAAPI entry points information).

```
cd ~/vaapi
git clone https://github.com/intel/libva-utils
cd libva-utils
./autogen.sh --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu
time make -j$(nproc) VERBOSE=1
sudo make -j$(nproc) install

```

At this point, issue a reboot:

`sudo systemctl reboot`

Then on resume, proceed with the steps below.

4. Build Intel's MSDK:

This package provides an API to access hardware-accelerated video decode, encode and filtering on Intel® platforms with integrated graphics. See the supported platform details below:

**Supported platforms:**

Compared to the open source VAAPI driver, this project supports the following SKU generations:

```
BDW (Broadwell)

SKL (Skylake)

BXT (Broxton) / APL (Apollolake)

CNL (Cannonlake)
```

Unless explicitly listed, any platform not in that list should be considered as unsupported. For these platforms, stick to upstream VAAPI.

**Build steps:**

(a). Fetch the sources into the working directory ~/vaapi:

```
cd ~/vaapi
git clone https://github.com/Intel-Media-SDK/MediaSDK msdk
cd msdk
git submodule init
git pull
```

(b). Configure the build with GCC:

(i). For Apollo Lake:

`perl tools/builder/build_mfx.pl --cmake=intel64.make.release --target=BXT`

(ii). For SKL:

`perl tools/builder/build_mfx.pl --cmake=intel64.make.release `

This will build MSDK binaries and MSDK samples.

(c ). Run the build:

`make -j$(nproc) -C __cmake/intel64.make.release`

(d). And install:

```
cd __cmake/intel64.make.release

sudo make -j$(nproc) install

```
Apply this workaround (needed by external dependencies such as FFmpeg's QSV encoder enablement):

```
sudo mkdir -p /opt/intel/mediasdk/include/mfx
sudo cp -vr /opt/intel/mediasdk/include/*.h  /opt/intel/mediasdk/include/mfx
```

Create a library config file for the iMSDK:

`sudo nano /etc/ld.so.conf.d/imsdk.conf`

Content:

```
/opt/intel/mediasdk/lib
/opt/intel/mediasdk/plugins
```

Then run:

`sudo ldconfig -vvvv`

To proceed.

Confirm that `/opt/intel/mediasdk/lib/pkgconfig/libmfx.pc` and `/opt/mediasdk/lib/pkgconfig/mfx.pc` are populated correctly:

```
Name: mfx
Description: Intel(R) Media SDK Dispatcher
Version: 1.26  ((MFX_VERSION) % 1000)

prefix=/opt/intel/mediasdk
libdir=/opt/intel/mediasdk/lib
includedir=/opt/intel/mediasdk/include
Libs: -L${libdir} -lmfx -lstdc++ -ldl -lva -lstdc++ -ldl -lva-drm -ldrm
Cflags: -I${includedir} -I/usr/include/libdrm
```

Just check. You don't need to edit the files.

When done, issue a reboot:

` sudo systemctl reboot`

Brainiarc7, 2018-04-21:
Confirm it fixes your issue.

closed 2019-08-27 by XinfengZhang