intel/media-driver #137: Supported platform

DavidMarec, 2018-04-10:
This project ' s description asserts that it supports Apollo Lake chipsets.
But, on the other hands, it requires `gmmlib` for its build and this one does not include this family in its summary.

-- See https://github.com/intel/gmmlib/issues/11 --

Is there any mismatch here ?

For example, I expect this driver  (by the fact the Intel msdk) to run on a board that embeds a E3950 processor.
If I m not been mislead, it ships a HD 505 graphic chip, so a 7Gen graphic product.

From:
* https://www.intel.com/content/www/us/en/architecture-and-technology/visual-technology/graphics-overview.html
* https://ark.intel.com/products/96488/Intel-Atom-x7-E3950-Processor-2M-Cache-up-to-2_00-GHz

I'm questioning about this because I m facing about 30% decreased performances between the Proprietary 's intel graphic stack and the open source one.

XinfengZhang, 2018-05-25:
@DavidMarec  sorry for the late response.  this driver should only support Gen9+. how about the whole stack. include open source msdk or just update to open source media driver. ... more detail information

DavidMarec, 2018-06-01:
I have been told that this is a mistake in the gmmlib description.

On the other hand, this page mentioned the HD505 as an 7Th GEN
https://www.intel.com/content/www/us/en/architecture-and-technology/visual-technology/graphics-overview.html

But it sounds it's a 9Th gen ' one:
here is what I get from i915_capabilities :
`
gen: 9
platform: BROXTON
pch: 0
is_mobile: no
is_lp: yes
is_alpha_support: no
has_64bit_reloc: yes
has_aliasing_ppgtt: yes
has_csr: yes
has_ddi: yes
has_dp_mst: yes
has_reset_engine: no
has_fbc: yes
has_fpga_dbg: yes
has_full_ppgtt: yes
has_full_48bit_ppgtt: yes
has_gmbus_irq: yes
has_gmch_display: no
has_guc: yes
has_guc_ct: no
has_hotplug: yes
has_l3_dpf: no
has_llc: no
has_logical_ring_contexts: yes
has_logical_ring_preemption: yes
has_overlay: no
has_pipe_cxsr: no
has_pooled_eu: yes
has_psr: no
has_rc6: yes
has_rc6p: no
has_resource_streamer: yes
has_runtime_pm: yes
has_snoop: yes
unfenced_needs_alignment: no
cursor_needs_physical: no
hws_needs_physical: no
overlay_needs_physical: no
supports_tv: no
has_ipc: yes
`

I am facing performance issue, https://software.intel.com/en-us/forums/intel-media-sdk/topic/779766

Any clue or knobs to activate into the i915 driver to speed up iHD  driver  ?
Adding HuC and GuC features doesn't improve anything on my side.