intel/media-driver #142: [gstreamer-vaapi][vpp] rotation doesn't work.

feiwan1, 2018-04-12:
System Environment
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver f6283e2
gst-vaapi tag 1.13.90

Reproduce Steps
$ LIBVA_TRACE=./rotation gst-launch-1.0 videotestsrc ! vaapipostproc ! vaapisink rotation=90

[rotation.194518.thd-0x00007d09.log](https://github.com/intel/media-driver/files/1901771/rotation.194518.thd-0x00007d09.log)
[rotation.194517.thd-0x00007d07.log](https://github.com/intel/media-driver/files/1901772/rotation.194517.thd-0x00007d07.log)
[rotation.194517.thd-0x00007d08.log](https://github.com/intel/media-driver/files/1901773/rotation.194517.thd-0x00007d08.log)

XinfengZhang, 2018-04-19:
@FurongZhang please take a look

feiwan1, 2018-04-19:
@sreerenjb , is there any feature/option in vaapipostporc are based on display attributes in libva/iHD?  If yes, the #141 may caused by this.

FurongZhang, 2018-05-15:
@feiwan1 @sreerenjb We checked the log, VPP based rotation is not called by GStreamer. Is it possible for you to take a look? Thank you!

XinfengZhang, 2018-06-07:
vaapi sink is base on vaputsurface which did not support rotation and other features. it was reported by caps, let's close this issue ,  if there are request please file a RFC to discuss it. thanks

closed 2018-06-07 by XinfengZhang