intel/media-driver #143: build error: ‘VA_STATUS_ERROR_NOT_ENOUGH_BUFFER’ was not declared in this scope

SpineEyE, 2018-04-14:
When I build with gcc 7.3.1 on Arch Linux, I get this error:

```
/home/xxx/builds/intel/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp: In member function ‘virtual VAStatus DdiEncodeBase::StatusReport(DDI_MEDIA_BUFFER*, void**)’:
/home/xxx/builds/intel/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:201:20: error: ‘VA_STATUS_ERROR_NOT_ENOUGH_BUFFER’ was not declared in this scope
             return VA_STATUS_ERROR_NOT_ENOUGH_BUFFER;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xxx/builds/intel/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:201:20: note: suggested alternative: ‘MOS_STATUS_NOT_ENOUGH_BUFFER’
             return VA_STATUS_ERROR_NOT_ENOUGH_BUFFER;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    MOS_STATUS_NOT_ENOUGH_BUFFER
[ 90%] Building CXX object media_driver/CMakeFiles/iHD_drv_video.dir/linux/common/codec/ddi/media_ddi_decode_avc.cpp.o
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video.dir/build.make:8103: media_driver/CMakeFiles/iHD_drv_video.dir/linux/common/codec/ddi/media_ddi_encode_base.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:164: media_driver/CMakeFiles/iHD_drv_video.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

XinfengZhang, 2018-04-16:
please update the libva, they just added a new feature, report the error of not enough bitstream buffer for encode in libva, https://github.com/intel/libva/pull/190

SpineEyE, 2018-04-16:
I see, it's not in 2.1.0, so that's actually a dependency mistake by the package maintainer. Sorry for bothering you.

closed 2018-04-16 by SpineEyE